suchmememanyskill/CYD-Klipper #8: FR: More info during print

Mordevolt, 2023-12-23:
Thank you for this project!

Is it possible to add more info on printing screen? It would be awesome if we can see all the info klipperscreen have: speed, flow, fan, model z high, etc...

suchmememanyskill, 2023-12-23:
I think displaying too much info will make the ui look very cluttered (i personally dislike how klipperscreen displays stats)

Maybe i can make some kind of menu button to show print stats

Mordevolt, 2023-12-23:
if it can be customizable - it would be perfect!

suchmememanyskill, 2024-02-03:
@Mordevolt can you test the following build for me:

https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7768066400
Install by downloading merged-firmware.bin from the link above, then flashing it in a commandline using esptool: `esptool write_flash 0x0 merged-firmware.bin`

During a print, the tab with the lighting icon (internally called the move tab) will have its contents replaced with a stats tab, that shows various statistics. If you think i missed any feel free to let me know

suchmememanyskill, 2024-02-05:
> Right, or really - what moonraker is emitting, right? I'll do some digging on my end as well and see if I can find the source & test a fix.

Yeah sorry, confusing mainsail with moonraker. Yes, moonraker provides me with the layer counts

https://moonraker.readthedocs.io/en/latest/printer_objects/#print_stats

matthawley, 2024-02-05:
I believe fluidd is parsing gcode on the startup (if I'm reading their codebase correctly) and simply looking for "layer" changes within. They also support `SET_PRINT_STATS_INFO` so I'll look at how that can be incorporated into Cura.

matthawley, 2024-02-05:
Okay, things are working after installing this preprocessor for Cura. https://github.com/pedrolamas/klipper-preprocessor

closed 2024-02-09 by suchmememanyskill