suchmememanyskill/CYD-Klipper #9: Wifi not working properly after  v1.1.2

Mordevolt, 2023-12-23:
After updating to v1.1.2 i noticed that wifi no linger working reliable. Wifi signal is good, but for some reason screen drops wifi connection ( i can see that via router statistics) and constantly trying to connect to Klipper (my record - 278), instead of wifi reconnect.
Mauby some kind of wifi connection check missing?
[image]

suchmememanyskill, 2023-12-24:
I haven't had any issues yet, but it's possible that me running code in the core that normally manages wifi may be causing issues. There are checks to see if the WiFi connection is intact, though: <https://github.com/suchmememanyskill/CYD-Klipper/blob/master/CYD-Klipper/src/ui/wifi_setup.cpp#L176>

Can you give me the logs when this screen is shown? (Connect it to a PC, open the installer page, then select the show logs option)

You can downgrade if you wish by grabbing the firmware combined .bin from the releases, and (assuming esptool is installed, and are in the same folder as merged-firmware.bin) typing `esptool write_flash 0x0 "merged-firmware.bin"` in cmd

suchmememanyskill, 2023-12-24:
Also, when the printer is on, is `http://{Klipper IP}/printer/info` accessible from your PC?

Mordevolt, 2023-12-24:
Please find logs attached.
Yes, printer/info accessible from my PC during this screen. Also, if i reboot devise via power its immediately cutch up connection.
The thing is that device completely drops wifi connection, i can confirm it via router logs, it just disconnecting from network and not reconnecting.
[esp-web-tools-logs.txt](https://github.com/suchmememanyskill/CYD-Klipper/files/13761993/esp-web-tools-logs.txt)

suchmememanyskill, 2023-12-24:
This is very very weird. I'm unsure if this is even my fault as i do check if the device is connected or not, so the wifi library is giving incorrect/inaccurate results then.

I have a suspicion on why this happens but i'll need some time to check. Please downgrade until then

Mordevolt, 2023-12-25:
I made downgrade to previous version, and noticed that it behaves a bit different: before "Connecting to Klipper" screen it displays "wifi connection" screen for a brief second, and resume operation. So probably the root case is something in Wi-Fi library

Mordevolt, 2024-01-07:
Did you have time to look into this? @suchmememanyskill

suchmememanyskill, 2024-01-07:
No.

I have 2 screens at this point, and neither show this behaviour

suchmememanyskill, 2024-01-07:
@Mordevolt can you check again with the latest dev branch build?
Relevant actions build for firmware files: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7440447897

Mordevolt, 2024-01-07:
just install dev build. will test it and post update here. Thank you!

Mordevolt, 2024-01-07:
seems this build fix the issue. Even when i manually disconnect CYD from router it reconnects almost instantly. Thank you again!

closed 2024-01-07 by suchmememanyskill