suchmememanyskill/CYD-Klipper #11: Offsets are calculated incorrectly in touchscreen_calibrate()

gknops, 2024-01-06:
In screen_driver.cpp, lines 76 and 79 it should be `10.0` instead of `20.0`

suchmememanyskill, 2024-01-06:
I derived it from [xtouch](https://github.com/xperiments-in/xtouch/blob/6b206baa31b4abe83da8c5e5295f2a827d31c6f6/src/devices/2.8/touch.h#L145), so that's wrong too then.

Why is it wrong and what's the effect of getting it wrong? Touch seems reasonably accurate still

gknops, 2024-01-06:
(BTW fabulous job on CYD_Klipper, thanks for making it available!)

I just happened to notice as I was copying bits from CYD_Klipper to get started with the CYD. The effect is small, but basically all touch events are shifted 10 pixels up and to the right. When you try with a stylus you will find that the bottom right of buttons does not work, but you can even click slightly outside to the top and left and a button will still work.

With that change everything will be spot-on.

gknops, 2024-01-06:
(Need recalibration after the change for the fix to show effect)

suchmememanyskill, 2024-01-06:
> CYD_Klipper to get started with the CYD

I am sorry for the mess that is my code, heh. I don't know c++, only c, and they are still quite different. Hopefully it'll still have some use to you, though

> are shifted 10 pixels up and to the right
Good to know. I can force a config wipe on update, maybe that's useful to do here

suchmememanyskill, 2024-01-06:
After testing, i uh, don't think it's wrong. It probably accounts for both sides being 10px off

gknops, 2024-01-06:
The two lines drawn meet at 10,10. However the offset calculation is done as if they were meeting at 20,20. I tested it in my ripped out code (works flawless after adjustment, off before), and I checked an unmodified CYD-Klipper (The "Screen Settings" button on the macro page is a good test case).

But I just realized that the same mistake was made in calculating `xDist` and `yDist` (should be `-20` instead of `-40`), so those errors kind of mitigate themselves a bit.

The crosses drawn are centered 10 pixels from the respective edges. So `xDist` and `yDist` should be screen width/height - 20.

Misterke, 2024-02-03:
I just had a look at the code of XTouch's touch.h and I indeed agree with @gknops that the calculation there is not using the correct coordinates for the center of the cross.  @gknops already added an issue on XTouch's github, but I'll be fixing it in my own version already!

suchmememanyskill, 2024-02-10:
@gknops https://github.com/suchmememanyskill/CYD-Klipper/commit/d3e7eec47a2c73633ca5164932833f1f25c06e22 should be fixed in this commit. Could you double check for me?

gknops, 2024-02-10:
Thanks! Looks good to me.

closed 2024-02-10 by suchmememanyskill