suchmememanyskill/CYD-Klipper #13: Remaining Time calculation

CrazyHairGuy, 2024-01-17:
Right now the remaining time seems to be a little bit off (by hours or more sometimes). I know that there are multiple options for estimating time in klipper (filament usage, file percentage, and my personal preference, slicer estimation) and was wondering if it would be possible to add an option to select which estimation method to display.

suchmememanyskill, 2024-01-17:
I currently do not fetch any slicer data at all, as Moonraker makes you do more api requests for that, and those are annoying/expensive to do

Currently it's based on file% and print time

CrazyHairGuy, 2024-01-18:
Where does it grab print time from? It doesn't seem to match any other numbers I can find.

suchmememanyskill, 2024-01-18:
https://moonraker.readthedocs.io/en/latest/printer_objects/#print_stats

"print_duration”

suchmememanyskill, 2024-02-18:
@CrazyHairGuy 

https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7948521490 Can you try this build? Grab the merged-firmware .bin from the github actions artifact firmware zip, install esptool and run esptool write_flash 0x0 merged-firmware.bin. This will wipe your settings.

In the build above a new option has been added to configure how estimated time is calculated:
- Percentage: Old behaviour (takes the file% and print time and calculates estimated time)
- Slicer: Takes the current print time and subtracts it from the slicers estimated seconds. If this time becomes negative (if the print takes longer than the slicer estimated) it will fall back on percentage based calculation
- Interpolated: Takes percentage and slicer estimates and mixes them. The weights are determined by file%, at the start of a print the time is fully estimated using the slicer estimates. The more the print goes on the more the percentage estimate is mixed in

Do you think this is good enough for your usecase?

closed 2024-02-21 by suchmememanyskill

CrazyHairGuy, 2024-02-21:
Sweet! I use [Klipper Estimator](https://github.com/Annex-Engineering/klipper_estimator) as a post process script, so the Slicer option is usually within 15-30 seconds of actual runtime. This should work perfectly.