suchmememanyskill/CYD-Klipper #19: Full Keyboard on Wifi Configuration

willsside, 2024-01-26:
Thanks for the great project! 

My wifi password has an "&" in the password. Would there be any way to add that symbol to the onscreen keyboard?

themacboy, 2024-01-26:
hahaha I have the same problem ! hahaha

suchmememanyskill, 2024-01-26:
Heh, the default lvgl keyboard i'm using doesn't include &, should be fixed bt upgrading the library

suchmememanyskill, 2024-01-26:
@themacboy @willsside could you try https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7670721311 ? You can install it by running the commandline tool esptool: `esptool write_flash 0x0 merged-firmware.bin`

willsside, 2024-01-26:
Just installed the updated firmware as you stated with esptool; new keyboard works great! Thanks for the quick turnaround!

themacboy, 2024-01-26:
Installed and working perfectlly !

razereqqq, 2024-01-31:
@suchmememanyskill I've got the same problem, except mine has an "^" in it, and even the updated firmware doesn't include that. I would greatly appreciate it If you could update it once again, changing that wifi password would be a major pain :/

suchmememanyskill, 2024-02-01:
@razereqqq fix available here: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7745111186

closed 2024-02-09 by suchmememanyskill