suchmememanyskill/CYD-Klipper #20: Toggle Wifi Power Adapter from Screen

N3m3515, 2024-01-30:
I am using a wifi Power Switch to toggle my Printer on and off from fluidd.
Is it Possible to implement the toggeling Feature into CYD-Klipper?
A third Option besides Firmware Resart and Restart would be good.

suchmememanyskill, 2024-01-30:
How does this work in fluidd? Does sending gcode still work or something? I can maybe show macros on the error screen

N3m3515, 2024-01-30:
It is defined as a power device in moonraker.conf

[image]

And it can be toggled via Macro but should be able to do so via monnraker api?

[image]

N3m3515, 2024-01-30:
I found it in the API Documentation:

https://moonraker.readthedocs.io/en/latest/web_api/#power-apis

suchmememanyskill, 2024-02-03:
@AndreyShpilevoy @N3m3515 can you both test the following build for me:

https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7768066400
Install by downloading merged-firmware.bin from the link above, then flashing it in a commandline using esptool: `esptool write_flash 0x0 merged-firmware.bin`

This build adds the ability to execute macros from the firmware not ready screen. Let me know if this is enough to turn the printer on with a macro (i do really hope klipper doesn't retrict the ability to turn on/off

suchmememanyskill, 2024-02-03:
[image]
oh my god, you seemingly are unable to, sigh

AndreyShpilevoy, 2024-02-03:
@suchmememanyskill I will check, I have a separate screen for it.
By the way, on my fork, I implemented it in a bit different way with REST API: https://moonraker.readthedocs.io/en/latest/web_api/#set-device-state

So in case it does not work with a webhook maybe it could be handled in such a way.
Of course, my implementation is SUPER ad hock and dirty, but it suits my needs.

Link to fork: https://github.com/AndreyShpilevoy/CYD-Klipper

suchmememanyskill, 2024-02-03:
Yeah i can indeed interact with the power api from moonraker, i just don't have any power devices to test anything with. So i cannot test my changes, which is pretty annoying

AndreyShpilevoy, 2024-02-03:
I can imagine.
As an option, you can simulate it.
You can create a power option based on Raspberry PI GPIO. And add LED or just measure the voltage there.

Here is some information: https://moonraker.readthedocs.io/en/latest/configuration/?h=%5Bpower#gpio-device-configuration

About testing:
If my printer is turned off I can see only "Printer error" or "Printer is not ready" screens with only 2 possible actions: "Restart" and "FW Restart". Not sure how I can try to call action from the screen.
But if I send a REST API request - the printer will react to it and will turn on the device (GPIO in my case that controls the cheapest solid-state relay).

Misterke, 2024-02-03:
> About testing: If my printer is turned off I can see only "Printer error" or "Printer is not ready" screens with only 2 possible actions: "Restart" and "FW Restart". Not sure how I can try to call [...]

This seems to be the case if your Klipper service is running even when the printer is off.  If you configure things correctly, then the Klipper service should only be started when the USB device of your printer is connected to the raspberry pi (or whatever you are using).  And if the Klipper service is not running, then Moonraker will correctly signal this as the printer being powered off.

I hope that helps.

AndreyShpilevoy, 2024-02-03:
@Misterke you are completely right.

If I add `bound_services: Klipper` to my `[power printer]` section it will handle it correctly.

Now I can see that the screen is on the 'Connecting to Klipper' page and has only the button "Reset". Also not ideal.
On my branch I added BTN with API POST call to both Screen - and Klipper error, and Connecting to Klipper.

I think it would be nice to have a setting, where we can provide the name of our device (needed for API calls) and show the "Turn ON" button if it's settled up.

suchmememanyskill, 2024-02-03:
@AndreyShpilevoy @N3m3515 

https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7769567201 build that actually implements it this time.

Power devices are shown as toggles in the Macros tab and in the printer error state screen there's a button called 'Devices'.
Please let me know if it works.

AndreyShpilevoy, 2024-02-03:
@suchmememanyskill works perfectly fine on `printer error state screen`.

Is it possible to add same button to `Connecting to Klipper` page to handle situations when the power device is bound with Klipper service?
Like here: https://docs.mainsail.xyz/overview/quicktips/printer-power-switch
If Klipper service is off CYD shows `Connecting to Klipper` with a single `Reset` btn.

suchmememanyskill, 2024-02-03:
@AndreyShpilevoy that screen is shown when i don't have access to moonraker. I do need at least moonraker to read the power devices

AndreyShpilevoy, 2024-02-03:
@suchmememanyskill can you try to turn off `Klipper` on the right top menu on your side?
Because on my side, when you turn off Klipper, moonraker is still available and responds on Device API calls

AndreyShpilevoy, 2024-02-03:
@suchmememanyskill f.e.
`http://192.168.68.118/printer/info` call will return:
`{"error": {"code": 503, "message": "Klippy Host not connected", "traceback": "Traceback (most recent call last):\n\n  File \"/home/pi/moonraker/moonraker/components/application.py\", line 699, in _process_http_request\n    result = await self.api_defintion.request(\n\n  File \"/home/pi/moonraker/moonraker/components/klippy_connection.py\", line 574, in request\n    raise ServerError(\"Klippy Host not connected\", 503)\n\nmoonraker.utils.ServerError: Klippy Host not connected\n\n\nThe above exception was the direct cause of the following exception:\n\n\nTraceback (most recent call last):\n\n  File \"/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1790, in _execute\n    result = await result\n\n  File \"/home/pi/moonraker/moonraker/components/application.py\", line 682, in get\n    await self._process_http_request(RequestType.GET)\n\n  File \"/home/pi/moonraker/moonraker/components/application.py\", line 703, in _process_http_request\n    raise tornado.web.HTTPError(\n\ntornado.web.HTTPError: HTTP 503: Klippy Host not connected\n"}}`

But `http://192.168.68.118/machine/device_power/devices` will return
`{"result": {"devices": [{"device": "printer", "status": "off", "locked_while_printing": false, "type": "gpio"}]}}`

So it still should be possible to connect to Moonraker even if Klipper is off.

suchmememanyskill, 2024-02-03:
@AndreyShpilevoy i do have to wonder in what situation realistically Klippy isn't available but Moonraker is. I suspect in a scenario where Moonraker is hosted on one machine and Klipper on another but this sounds out of scope for this issue.

AndreyShpilevoy, 2024-02-03:
@suchmememanyskill my situation :)
Hosted on the same machine. 

My config - RPI powered directly from the socket, Printer PSU can be controlled by a solid-state relay.
If you turn off power via Moonraker - the relay will shut down the PSU and printer board. RPI still running.
Also, in Moonraker.conf you can set up `[power printer]` with `bound_services: klipper`. It will shut down Klipper on PRI when your Device is in an OFF state. Then Klipper will not show any error and you will see a very nice picture on your web interface.

Like here: https://docs.mainsail.xyz/overview/quicktips/printer-power-switch
[image]

Otherwise, if you don't bind your Klipper service with the `[power printer]` toggle - it will throw Connecting errors.

suchmememanyskill, 2024-02-03:
Alright, finally starting to grasp your issue @AndreyShpilevoy

https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7769864071

Should hopefully be the last one :)

AndreyShpilevoy, 2024-02-03:
@suchmememanyskill yay!
Thank you a lot! It's exactely what I want!
Now I can get rid of my ugly ad-hock branch :)

Thank you for the fast reaction, responsivness and nice communication :)
And for such a dreat aproject as well 👍

N3m3515, 2024-02-04:
Hello, the last Firmware works but i only get the Devices button after Power Cycling the Printer once. The button is missing from the Printer not ready Screen after i start the CYD.

suchmememanyskill, 2024-02-04:
That sounds really odd, i do hope the power devices can be queried when another call to moonraker succeeds

closed 2024-02-09 by suchmememanyskill