suchmememanyskill/CYD-Klipper #22: Multiple Printers on one Instance

WinstonLeWolf, 2024-02-01:
I have a Dell computer running Klipper that controls multiple printers.  The whole Klipper instance has one IP address and each printer has separate ports.  When I setup my screen, I enter the IP address but if I enter any port other than the default 80, it fails to connect.  It is possible to view another printer besides the one that's on the first port of my Klipper instance?

suchmememanyskill, 2024-02-01:
Is moonraker running on those ports?

WinstonLeWolf, 2024-02-01:
> Is moonraker running on those ports?

Yep.  Each printer has it's own instance of Moonraker.

suchmememanyskill, 2024-02-01:
Is the moonraker api accessible from the network, like `http://{ip}:80/server/info`?

WinstonLeWolf, 2024-02-01:
> Is the moonraker api accessible from the network, like `http://{ip}:{port}/server/info`?

It says the IP refused to connect.  Is there a way to change that?  Right now in order to select a different printer, I go to the IP address, then use the dropdown to select each printer in the Mainsail Interface.

suchmememanyskill, 2024-02-01:
I don't know anything about multiple printer same device setups, sorry

WinstonLeWolf, 2024-02-01:
> I don't know anything about using multiple printer same device setups, sorry

Thanks for trying.  Keep up the good work.

WinstonLeWolf, 2024-02-02:
> > I don't know anything about using multiple printer same device setups, sorry
> 
> Thanks for trying. I'll do some research on making that port visible and update the post if I find anything.

I think I found the solution.  You have to add

enable_api_key: True

to moonraker.conf in order for the connection to occur.  Hopefully this helps someone else

closed 2024-02-02 by WinstonLeWolf