suchmememanyskill/CYD-Klipper #23: Z-Offset

AndreySemjonov, 2024-02-02:
Would be great if there would be separate built in Z-Offset tab with buttons same as we have in mainsail (or fluid etc).
Try to make it with macros for now, but can't get it working so far.

suchmememanyskill, 2024-02-03:
Why would you need to constantly change your Z offset?

AndreySemjonov, 2024-02-03:
Not constantly, but its needed from time to time.
For example I have double sided PEI sheet (smooth and textured). And each side needs their own Z height (I use beacon).
I don't like tinkering Z height with paper etc. So instead of this I am just making bigger skirt (when print 1st on each side) and when do print I adjust Z offset so layers looking good.
After that just save settings.

suchmememanyskill, 2024-02-03:
@AndreySemjonov can you test the following build for me:

https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7768066400
Install by downloading merged-firmware.bin from the link above, then flashing it in a commandline using esptool: `esptool write_flash 0x0 merged-firmware.bin`

During a print, the tab with a lighting icon will be replaced with a statistics tab where you can also set the z offset offset(?)

AndreySemjonov, 2024-02-03:
> @AndreySemjonov can you test the following build for me:
>
> https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7768066400 Install by downloading merged-firmware.bin from the link above, [...]

Nice thanks. I will test it soon

AndreySemjonov, 2024-02-04:
> @AndreySemjonov can you test the following build for me:
>
> https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7768066400 Install by downloading merged-firmware.bin from the link above, [...]

I tested. It's working. Thank you!!!

closed 2024-02-09 by suchmememanyskill