suchmememanyskill/CYD-Klipper #25: [FR] OTA Update

Misterke, 2024-02-03:
I know the README even mentions there is no OTA support, but I don't understand why there isn't.  XTouch has OTA support and that just requires some place on the Internet hosting the bin file and a JSON version file, so why force everyone through manual updates?

If there is any good reason why OTA updating would not be possible, then at least clearly mention that reason in the README instead of just mention that it isn't included.

suchmememanyskill, 2024-02-03:
@Misterke i don't know how it works, that's the simple answer. If you can point me to how it works in xtouch's repo, i'd be very grateful

Misterke, 2024-02-03:
> @Misterke i don't know how it works, that's the simple answer. If you can point me to how it works in xtouch's repo, i'd be very grateful

Well, I have not dug completely into how xtouch does it, but quickly browsing through the code, I'd say:

- Most of the needed code is in https://github.com/xperiments-in/xtouch/blob/main/src/xtouch/firmware.h, function xtouch_firmware_checkOnlineFirmwareUpdate
- It first downloads http://xperiments.in/xtouch-bin/ota/ota.json to the sdcard, parses that JSON file and extracts the version stored in it
- If that version is higher than the one of the currently running firmware, it downloads the firmware URL mentioned in that JSON file to the sdcard
- When the download is complete it verifies the MD5 hash of the file on the sdcard with that mentioned in the JSON file
- If also that seems OK, it triggers a reboot (I guess to ensure there is no other activity that could hamper the upgrading)
- At startup the function xtouch_firmware_checkFirmwareUpdate is executed and that basically does the same checks again, but with the files stored on the sdcard and if it also determines that the file on the sdcard is a valid update, then it uses the Update object from the Update library of the ArduinoEspressif32 framework and feeds the file to it.   When the upgrade is done (failed or succeeded) it then deletes the JSON and firmware.bin files and triggers another reboot.  If the upgrade succeeded, that should then boot the new version.

I hope that helps and would allow you to add something similar to the Klipper version.  I'm still awaiting an extra CYD for use for Klipper and haven't even looked at your code yet, so I have no idea whether it is similar to how the xtouch code is structured.  If it is, then adding the same mechanism could be pretty easy.

suchmememanyskill, 2024-02-03:
@Misterke Do note that for cyd-klipper, no sd card is needed. This kind of hampers my ability to do this as the firmware package is larger than all available system memory

N3m3515, 2024-02-04:
why not add a check if an sd card is present and if so display an ota button and use the sdcard for that?

suchmememanyskill, 2024-02-04:
> why not add a check if an sd card is present and if so display an ota button and use the sdcard for that?

Because i don't want to require additional hardware for any feature

N3m3515, 2024-02-04:
fair enough

dronecz, 2024-02-05:
There are some projects, which are based on GitHub releases. To name a few: 

- [esp_ghota](https://github.com/Fishwaldo/esp_ghota)
- [ESP32-OTA-Pull](https://github.com/mikalhart/ESP32-OTA-Pull)
- [esp32-flashz](https://github.com/vortigont/esp32-flashz)

Maybe something like this could be implemented and not require additional hardware.

suchmememanyskill, 2024-02-09:
Ota hopefully implemented in https://github.com/suchmememanyskill/CYD-Klipper/commit/64266b1ff83683fd4f2198f9cce34f08afe4acbe

An update button will appear in the settings if there's an update. There's also a toggle for automatic updates, that will update if needed right after connecting to wifi

closed 2024-02-10 by suchmememanyskill