suchmememanyskill/CYD-Klipper #26: Question + thanks

szafranski, 2024-02-03:
Hi! 

First of all - thank you for awesome project. Second - is there any way to install newer/non-official builds like from dev branch or so? Net installer points strictly to one release.

Thanks!

suchmememanyskill, 2024-02-03:
Ya, you can grab builds from my dev branches from the github actions tab: https://github.com/suchmememanyskill/CYD-Klipper/actions

Install by downloading merged-firmware.bin from the artifacts from one of the successful builds, then flash it in a terminal/command prompt using esptool: `esptool write_flash 0x0 merged-firmware.bin` (you need to install esptool if you haven't already)

szafranski, 2024-02-03:
Thank you! Is there thumbnails preview on the roadmap maybe? Sorry if bothering

suchmememanyskill, 2024-02-03:
> Thank you! Is there thumbnails preview on the roadmap maybe? Sorry if bothering
There's an experimental branch that has it but it's unstable as parsing images takes a lot of memory. I don't really see it happening with the little memory i have

szafranski, 2024-02-03:
Understood. Thank you for great work.

closed 2024-02-03 by suchmememanyskill