suchmememanyskill/CYD-Klipper #27: Support hostname for klipper

matthawley, 2024-02-03:
Could you support adding your klipper instance by hostname and not just IP? I've went ahead and made my IP static (I think) but it would be much easier to reference it by hostname (but also still support IP).

suchmememanyskill, 2024-02-03:
In my experience with networking hostname is super inconsistent. I'd rather not.

Can't you uh, set up your router's default dns to include an entry for your machine?

matthawley, 2024-02-04:
I'd argue otherwise - my IP is inconsistent if using DHCP (the default for wireless routers) but my hostname of my raspberry pi (or any other computer) is static. My router does provide internal DNS for hostnames, but this install ONLY allowed me to enter my instance by IP (unless I was completely blind and couldn't find a way to switch away from numeric input).

matthawley, 2024-02-04:
Looking through the code, it seems you're just making an http request through an http client, which should work with a hostname. It might be as simple as not setting the keyboard mode (or making numeric entry default)?

https://github.com/suchmememanyskill/CYD-Klipper/blob/2a2fff27d6db0e4ba1d25cecd0901ebe72196511/CYD-Klipper/src/ui/ip_setup.cpp#L111

suchmememanyskill, 2024-02-04:
@matthawley could you fork it and try it? (Github actions should automatically produce a new binary for you, you can flashed the merged firmware bin. I've explained how to flash it in the keyboard characters issue)

matthawley, 2024-02-04:
@suchmememanyskill will do! Is this the correct documentation for the keyboard? If so, we could probably create a custom keyboard to only include `a-z`, `0-9` and `.` characters

suchmememanyskill, 2024-02-04:
> @suchmememanyskill will do! Is this the correct documentation for the keyboard? If so, we could probably create a custom keyboard to only include `a-z`, `0-9` and `.` characters
> 
> https://docs.lvgl.io/master/widgets/keyboard.html#overview

I'm using lvgl v8, but looks like it. i have a fork set up for keyboard changes if needed

matthawley, 2024-02-04:
Added updated keyboard in #28

closed 2024-02-09 by suchmememanyskill