suchmememanyskill/CYD-Klipper #30: CYD Boot Loops after connecting to Mainsail

Gremlin1987, 2024-02-05:
I have two CYD units, after configuring the WIFI and Mainsail IP address they both start too boot loop, the log seems to indicate it's as soon as they connect to the web host.

I have attached the log for reference.

[esp-web-tools-logs.txt](https://github.com/suchmememanyskill/CYD-Klipper/files/14159028/esp-web-tools-logs.txt)

suchmememanyskill, 2024-02-05:
@Gremlin1987 i can't get the required info out of the logs.

Would you mind loading up cyd klipper in visual studio code + platformio (same folder with the platformio.ini file), then pressing the upload & monitor workflow, then posting the extended logs of that here?

Gremlin1987, 2024-02-05:
Hello,

This may have been an issue with my OrangePi/mainsail. I used visual studio per your request to upload 1.14 to one of my CYDs and after doing so it was able to connect properly. When I powered on the CYD I had set up with the web installer yesterday it also connected without issue.

Yesterday was the first day I ever had Klipper up and running on my printer and control over web browser worked fine. I shutdown the OrangePi over night and that seems to have allowed the CYD to connect as well.

Thank you for the quick support!

closed 2024-02-05 by suchmememanyskill