suchmememanyskill/CYD-Klipper #40: Display Layers stay at 0 / 0 during print.  BUG?

m00x-pixel, 2024-02-11:
Hi,

Thank you for you great work, i have updated to to the latest version 1.3.0 it was easy as always. 
Maybe there is some Bug.

At second screen during a print the layer info stay always at 0 / 0

[image]

Within Fluuid and Mainsail it show the layers like  "Layer 24 / 128" and counting.
Is there anything i need to set or is it not  read the information correctly from moonraker? 

Greetings

suchmememanyskill, 2024-02-11:
https://moonraker.readthedocs.io/en/latest/printer_objects/#print_stats (see 'info')

https://docs.mainsail.xyz/overview/slicer/prusaslicer#layer-count

If those commands aren't ran (and thus no accurate layer info is available), Mainsail/Fluidd guess the layer count based on the layer height, but this isn't fully accurate (you'll see this if the z compensation is high enough the current layer count becomes unstable).

Cyd-klipper doesn't do this guesswork, and just uses the accurate layer info if available. If there's no info available, it stays at 0.

m00x-pixel, 2024-02-11:
Thank you for the clarification, I understand what I have overlooked here.
So the issue can be closed.

closed 2024-02-11 by suchmememanyskill