suchmememanyskill/CYD-Klipper #44: Bug: Invert Colors option is inverted (on when turned off)

aamott, 2024-02-14:
It looks like the "Invert Colors" option is inverted. When switched off, blue shows as yellow and etc. for other colors, and light/dark mode is inverted as well. 

Not gonna lie, I think I like the inverted version of light mode with yellow (alternative dark mode blue) more than the proper dark mode with blue, and I probably wouldn't have found out without this bug, so I'm pretty happy, but still good to report :)

suchmememanyskill, 2024-02-14:
This isn't really a bug i can fix. The CYD variants with a usb-c port have inverted screens

aamott, 2024-02-17:
Ah, makes enough sense. I'll close the issue then.
Any chance you'd add a page documenting the variants of CYD? I looked online and didn't find anything specific. I bet there would be support if the variants could if the variants could be listed somewhere. I'd be happy to contribute anything I learn if you do! Honestly, I'd even throw something together with the little bit I can find if you'd host it here. I just don't want to create a separate place with the info and fragment the information.

closed 2024-02-17 by aamott

suchmememanyskill, 2024-02-18:
https://github.com/rzeldent/platformio-espressif32-sunton

This repo has documentation on all cyd variants. There's an experimental driver implementing these definitions so the screen won't be inverted by default. (Also, the colors are way better when using the actual right definition for a screen, not some hack that targets all 3 variants)

I have considered releasing versions for all 3 versions, but to be honest this sounds like a nightmare as people can easily install the wrong one by accident as the names/models are so similar