suchmememanyskill/CYD-Klipper #46: Support for Probe /

yourbuddydinec, 2024-02-15:
Hey there. 
Was hoping it could one day be possible to add support for the probe calibration? Would help a lot since the main thing I use the screen at the printer for is so I can probe calibrate there with one hand while using a shim with the other. I believe Probe / Beacon / Cartographer / Eddy all use the same calibration UI from within klipper.

[image]

yourbuddydinec, 2024-02-15:
Its the z offset / probe calibration. So you do this to find out how far away the nozzle is from the probe so that the probe is useful. The one built in one is useful though not a ton since i had THOUGHT the numbers only went down to 0.1 and not 0.01 or 0.005 like it shows in the previously attached screenshot. 
It also HAS to be done BEFORE you print for probes like the cartographer or beacon or other eddy current probes so it knows how far away of a reading the bed should be since it can have numbers WILDLY different. Like the screenshot above shows it starting with a number saying your probe is 5.291mm away from the nozzle and this one in the screenshot below for an eddy current probe has a reading of 403. 403 what? No idea. Doubtful its mm but its a reading of 403.
[image]

suchmememanyskill, 2024-02-15:
I don't think i 100% get what the ui in the image is for, exactly.

https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7919973757 Can you try this build? Grab the merged-firmware .bin from the github actions artifact firmware zip, install esptool and run `esptool write_flash 0x0 merged-firmware.bin`. This will wipe your settings.

This allows you to access the 'Z-Offset' setter in CYD-Klipper via (move menu, charge symbol) > (Home if steppers are unlocked) > Params > Z Offset. Is this sufficient for your use case?

Misterke, 2024-02-15:
@suchmememanyskill , the screenshot shown is what Mainsail shows in response to a Klipper `PROBE_CALIBRATE` or `MANUAL_PROBE` command.  When issuing those the Klipper parser basically enters a separate "mode" where the extra commands  `TESTZ Z=<value>`, `ACCEPT` and `ABORT` allow interactive tuning of the Z offset.

I have used those myself in the past, but honestly find that it is easier to just home, move the nozzle to the position where the probe homed and Z = 0.1mm and then use the paper test while just dynamically adjusting the Z offset (I do it mostly in Mainsail until now).

So I guess the requester is asking for the addition of such a dedicated GUI as Mainsail has in that special mode, but I think Mainsail detects that mode via parsing the console messages, which I guess might be beyond what a tiny ESP32 could handle ...

yourbuddydinec, 2024-02-15:
The firmware did flash just fine and the menu almost works. Normally what the probe interface does is when you click the amounts (0.005, 0.01, 0.05, 0.1, 1 and -0.005, -0.01, -0.05, -0.1, -1) it moves the nozzle that amount so that you can then move the shim or paper under it to see if you feel resistance. Once you find a little resistance you just click accept and it then saves that z offset and uses it at that point.

The firmware seemed to allow you to change what the z offset is but doesn't move at the same time to let you check if that z offset its using is correct using a shim or paper. Firmware also let you then move the nozzle to z=0 or z=1 with the currently changed z offset from the screen as far as I can tell.

I mean I guess honestly just doing a `SET_GCODE_OFFSET z={amount} MOVE=1` of those amounts in a macro and assigning them to the macro section could work too but you would need 10 macros total. I was able to replicate Mainsail / Fluidd's version ALMOST perfectly with this macro
```
SET_GCODE_OFFSET Z_ADJUST=1.0 MOVE=1
SET_GCODE_OFFSET Z_ADJUST=-1.0 MOVE=1
SET_GCODE_OFFSET Z_ADJUST={negative move amount} MOVE=1
```
but i did notice if the move amount moved it down below your z_min it would do the hop but then not do the final move making it LOOK like it didn't do anything since it basically just moved back to the same place and didn't do the final move. The upwards 1.0 and downwards -1.0 are only needed for negative amounts since it seems to be to give you time to get your shim/paper in place and you don't really need that if moving upwards. The little hop is built into the normal version and is useful so thought I would include it.

suchmememanyskill, 2024-02-16:
@yourbuddydinec 
> The firmware seemed to allow you to change what the z offset is but doesn't move at the same time to let you check if that z offset its using is correct using a shim or paper. Firmware also let you [...]

But that is um, exactly what it's doing. https://github.com/suchmememanyskill/CYD-Klipper/blob/dev/CYD-Klipper/src/ui/panels/stats_panel.cpp#L49
(Do note here it only adds MOVE=1 when the printer is idle, which is seemingly incorrect looking at mainsail's source, it should always be applied). If MOVE was not there changing the z offset would change the Z height at the same time.

Are the increments so small you cannot tell? I couldn't visually tell until i spammed the highest + increment to 1mm z offset.

suchmememanyskill, 2024-02-17:
@yourbuddydinec 
I have fixed the MOVE=1 situation in the following commit: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7941271385/job/21683439475

Also, this swaps out the ajustment param 0.005 for 0.2 (i feel like there's very little situations where you need that kind of precision).

With the 0.2 option i can visibly see my printer move

closed 2024-02-21 by suchmememanyskill