suchmememanyskill/CYD-Klipper #48: "Failed to connect": endpoint returning 401

flaviut, 2024-02-19:
When I put in my printer's details in setup, I get a "Failed to connect" error. This is [because CYD-Klipper makes a request to `/printer/info`](https://github.com/suchmememanyskill/CYD-Klipper/blob/a436c6b5c7a377ac65a6364b2cc5d6363e6fd1e5/CYD-Klipper/src/ui/ip_setup.cpp#L34), which then fails with a 401:

```json
{
  "error": {
    "code": 401,
    "message": "Unauthorized",
    "traceback": "Traceback (most recent call last):\n\n  File \"/home/user/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1769, in _execute\n    result = await result  # type: ignore\n\n  File \"/home/user/moonraker/moonraker/components/application.py\", line 1047, in prepare\n    await ret\n\n  File \"/home/user/moonraker/moonraker/components/application.py\", line 502, in prepare\n    self.current_user = await auth.authenticate_request(\n\n  File \"/home/user/moonraker/moonraker/components/authorization.py\", line 843, in authenticate_request\n    raise HTTPError(401, \"Unauthorized\")\n\ntornado.web.HTTPError: HTTP 401: Unauthorized (Unauthorized)\n"
  }
}
```

It would be nice to be asked for an API key so that these requests succeed.

<details>
<summary>For now, I've worked around this with the following patch</summary>

```diff
diff --git a/CYD-Klipper/src/core/data_setup.cpp b/CYD-Klipper/src/core/data_setup.cpp
index d240d6d..8c10f4d 100644
--- a/CYD-Klipper/src/core/data_setup.cpp
+++ b/CYD-Klipper/src/core/data_setup.cpp
@@ -49,6 +49,7 @@ void send_gcode(bool wait, const char *gcode)
     sprintf(buff, "http://%s:%d/printer/gcode/script?script=%s", global_config.klipperHost, global_config.klipperPort, urlEncode(gcode).c_str());
     HTTPClient client;
     client.begin(buff);
+    client.addHeader("X-Api-Key", "aaaaa");
 
     if (!wait)
     {
@@ -73,6 +74,7 @@ void fetch_printer_data()
     HTTPClient client;
     client.useHTTP10(true);
     client.begin(buff);
+    client.addHeader("X-Api-Key", "aaaaa");
     int httpCode = client.GET();
     delay(10);
     if (httpCode == 200)
diff --git a/CYD-Klipper/src/core/files_query.cpp b/CYD-Klipper/src/core/files_query.cpp
index 0fce127..3a67563 100644
--- a/CYD-Klipper/src/core/files_query.cpp
+++ b/CYD-Klipper/src/core/files_query.cpp
@@ -33,6 +33,7 @@ FILESYSTEM_FILE* get_files(int limit){
     client.useHTTP10(true);
     client.setTimeout(5000);
     client.begin(buff);
+    client.addHeader("X-Api-Key", "aaaaa");
     int httpCode = client.GET();
     auto timer_parse = millis();
 
diff --git a/CYD-Klipper/src/core/macros_query.cpp b/CYD-Klipper/src/core/macros_query.cpp
index 2ed0c0c..60e5b7b 100644
--- a/CYD-Klipper/src/core/macros_query.cpp
+++ b/CYD-Klipper/src/core/macros_query.cpp
@@ -18,6 +18,7 @@ static void _macros_query_internal(){
     HTTPClient client;
     client.useHTTP10(true);
     client.begin(url.c_str());
+    client.addHeader("X-Api-Key", "aaaaa");
     int httpCode = client.GET();
     if (httpCode == 200){
         JsonDocument doc;
@@ -48,6 +49,7 @@ void _power_devices_query_internal(){
     client.useHTTP10(true);
     client.setTimeout(500);
     client.begin(url.c_str());
+    client.addHeader("X-Api-Key", "aaaaa");
     int httpCode = client.GET();
     if (httpCode == 200){
         JsonDocument doc;
@@ -85,6 +87,7 @@ bool set_power_state(const char* device_name, bool state) {
     HTTPClient client;
     client.useHTTP10(true);
     client.begin(url.c_str());
+    client.addHeader("X-Api-Key", "aaaaa");
     if (client.POST("") != 200)
         return false;
 
diff --git a/CYD-Klipper/src/ui/ip_setup.cpp b/CYD-Klipper/src/ui/ip_setup.cpp
index 79887d8..e075edb 100644
--- a/CYD-Klipper/src/ui/ip_setup.cpp
+++ b/CYD-Klipper/src/ui/ip_setup.cpp
@@ -37,6 +37,7 @@ bool verify_ip(){
         Serial.println(url);
         client.setTimeout(500);
         client.begin(url.c_str());
+        client.addHeader("X-Api-Key", "aaaaa");
         httpCode = client.GET();
         return httpCode == 200;
     }
diff --git a/CYD-Klipper/src/ui/panels/print_panel.cpp b/CYD-Klipper/src/ui/panels/print_panel.cpp
index eeeb589..fb045a4 100644
--- a/CYD-Klipper/src/ui/panels/print_panel.cpp
+++ b/CYD-Klipper/src/ui/panels/print_panel.cpp
@@ -37,6 +37,7 @@ static void btn_print_file(lv_event_t * e){
 
     HTTPClient client;
     client.begin(buff);
+    client.addHeader("X-Api-Key", "aaaaa");
     int httpCode = client.POST("");
     Serial.printf("Print start: HTTP %d\n", httpCode);
 }
```
</details>

suchmememanyskill, 2024-02-19:
Can you not whitelist ip's in moonraker? It sounds very inconvenient to enter a long api key on a tiny screen

suchmememanyskill, 2024-02-19:
Actually, if it returns 401 i can just show a screen, however inconvenient that may be. I guess it beats opening a config file on the remote, though. Let me whip up a quick prototype

flaviut, 2024-02-19:
I have philosophical objections to IP whitelisting, but my workaround also works fine for me. Please don't spend time on my self-inflicted restrictions unless you want to!

suchmememanyskill, 2024-02-19:
@flaviut 

Would you mind trying the following build? https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/7965078098

If it detects a 401 from the moonraker api when trying to fetch /printer/info, it'll bring up an api key entry screen

flaviut, 2024-02-19:
Confirmed that the esp32-3248S035C version worked perfectly! While it was pretty easy, I now realize that this would be painful on a 2.4in display.

closed 2024-02-21 by suchmememanyskill