suchmememanyskill/CYD-Klipper #53: Allow Hidden WIFI SSID

tsk-2222, 2024-02-25:
I know this is a work in progress. I just wanted to suggest to allow the user to enter a hidden WIFI SSID.

suchmememanyskill, 2024-03-26:
@tsk-2222 can you try the following build:

Firmware: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8441544262
How to run a dev build: https://github.com/suchmememanyskill/CYD-Klipper/blob/dev/Build.md

tsk-2222, 2024-03-26:
When I manually type the SSID in (hidden SSID or visible)  it stalls at "Connecting  to WIFI". When I pick the SSID from the list it connects fine.

suchmememanyskill, 2024-03-27:
Can you look at the console output (either via visual studio code or the web installer)?

When i tested it, it worked fine

tsk-2222, 2024-03-27:
Done.

Wifi Status Disconnected <repeats> on console 
I type in hidden SSID and password
SSID shows up on console along with '>' command prompt
"Connecting to WIFI" displays on CYD 2.8"

The same thing happens with a different visible SSID when typed in manually. 
The keyboard isn't sanitizing underscores is it?

With the SSID and password saved, I get this after a reboot ("Connecting to WIFI" on screen):
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Hello World
Config version: 5
Screen init done
```

Is there a debug mode that would show more output?

suchmememanyskill, 2024-03-27:
This same build should have a feature that if you try to reconnect to the same SSID, you get a message asking you if you want to reuse the stored password. Do you get this popup?

Also, i expect to see more logs than this, especially https://github.com/suchmememanyskill/CYD-Klipper/blob/dev/CYD-Klipper/src/ui/wifi_setup.cpp#L119

To get more reliable logs, i don't think you have any other options than to set up a developer environment (visual studio code + platformio). The build doc walks you through it

tsk-2222, 2024-03-27:
> This same build should have a feature that if you try to reconnect to the same SSID, you get a message asking you if you want to reuse the stored password. Do you get this popup?
Yes I do

> Also, i expect to see more logs than this, especially https://github.com/suchmememanyskill/CYD-Klipper/blob/dev/CYD-Klipper/src/ui/wifi_setup.cpp#L119
> 
I'm not sure what's going on with the logs.  The console will stop displaying at "screen init done'"  but then if I hit "RESET DEVICE" the rest of the log will populate and I will see password length entry, OTA update results, etc. Also the device will reset  ofc.

Using the reset 'trick' I managed to get this entry in the log:
`Connecting to ���?���?���? with a password length of 13`

BTW, I am testing this with a visible SSID and I am just entering the SSID manually rather than picking from the list.

suchmememanyskill, 2024-03-27:
Yep, that's some corrupted text alright.
Found the bug using this; it's cause i don't copy the string, so the string eventually gets deallocated (inbetween ssid accept and password entry) and causes this.

https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8455372710 this build should fix that

tsk-2222, 2024-03-27:
It seems to be working. I tried manual entry with hidden and non-hidden SSID.  

At one point I did see panic error and had an unrecoverable boot loop. It happened after I typed in the IP address and port.   I cannot duplicate it though and I don't have a log. :(  It's possible I mistyped the numbers.

There does seem to be a separate issue with the device rebooting when it is connected to the server but the printer itself is 'not ready'.

Thanks for all the work!

closed 2024-03-27 by tsk-2222