suchmememanyskill/CYD-Klipper #54: ESP32-8048S043 screen

themacboy, 2024-02-27:
Good morning,

I made a mistake and purchased a 4.3inch ESP32-8048S043 screen.

This is a capacitative IPS RGB screen with 800*480 pixels and 8m PSRAm and 16M Flash memory.

It is possible to uset with your code?

suchmememanyskill, 2024-02-27:
https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8070991249

Can you try this build? Grab the merged-firmware .bin from the github actions artifact firmware zip, install esptool and run `esptool write_flash 0x0 merged-firmware.bin`

themacboy, 2024-02-28:
Not lucky.

```
hemacboy@themacboy-HP:~$ esptool write_flash 0x0 /home/themacboy/Escriptori/esp32-8048S043C-smartdisplay/merged_firmware.bin 
esptool.py v2.8
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type...
/dev/ttyUSB0 failed to connect: Unexpected UART datecode value 0x03700000. Failed to autodetect chip type.
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyS0 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 2 available serial ports.

```

suchmememanyskill, 2024-02-28:
You are holding down the boot button when it's attempting to connect, right?

Otherwise you can try to upload from visual studio code + the platformio plugin, but that's a lot trickier to set up

themacboy, 2024-02-28:
[image]

[image]

themacboy, 2024-02-28:
> 
> Otherwise you can try to upload from visual studio code + the platformio plugin, but that's a lot trickier to set up

ummm will read how to do that. Thx

themacboy, 2024-02-28:
This is from platformio. Screen is connected as /dev/ttyUSB0

[image]

I will try to lear how to compile in platformio

suchmememanyskill, 2024-02-28:
It's not too difficult.
Go to this url: https://github.com/suchmememanyskill/CYD-Klipper/tree/4.3inch

[image]
1. Download the code
2. Open the CYD-Klipper folder inside visual studio code

[image]
3. Open the platformio tab
4. Click on 'Upload'. It should eventually start flashing the screen (or the console will say connecting...), at that point the boot button should be held

themacboy, 2024-02-28:
Yes, I am successful in building and uploading.
But on the reboot screen it turns white and nothing else hehehe we've made another step forward

themacboy, 2024-02-28:
Curiously ... Im monitoring activity on board from platformio and looks like there is any activity on the background hehehe

[image]

themacboy, 2024-02-28:
Building log (looks very correct to me:

```*  Executing task: platformio run --environment esp32-8048S043C-smartdisplay 

Processing esp32-8048S043C-smartdisplay (board: esp32-8048S043C-smartdisplay; platform: espressif32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/suchmememanyskill/esp32-smartdisplay
git version 2.43.0
S'està clonant a «/home/themacboy/.platformio/.cache/tmp/pkg-installing-84hrw3lq»...
remote: Enumerating objects: 54, done.
remote: Counting objects: 100% (54/54), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 54 (delta 12), reused 24 (delta 4), pack-reused 0
S'estan rebent objectes: 100% (54/54), 1.74 MiB | 6.68 MiB/s, fet.
S'estan resolent les diferències: 100% (12/12), fet.
S'ha registrat el submòdul «boards» (https://github.com/rzeldent/platformio-espressif32-sunton.git) per al camí «boards»
S'està clonant a «/home/themacboy/.platformio/.cache/tmp/pkg-installing-84hrw3lq/boards»...
remote: Enumerating objects: 682, done.        
remote: Counting objects: 100% (446/446), done.        
remote: Compressing objects: 100% (146/146), done.        
remote: Total 682 (delta 416), reused 311 (delta 300), pack-reused 236        
S'estan rebent objectes: 100% (682/682), 51.06 MiB | 5.48 MiB/s, fet.
S'estan resolent les diferències: 100% (467/467), fet.
Camí de submòdul «boards»: s'ha agafat «4aa429fdefeda907a9e3e8d9b0f4a22feda4d152»
Library Manager: esp32_smartdisplay@2.0.5+sha.4e689b5 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing lvgl/lvgl @ ^8.3.9
Unpacking  [####################################]  100%          
Library Manager: lvgl@8.3.11 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ ^7.0.0
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@7.0.3 has been installed!
Library Manager: Installing plageoj/UrlEncode @ ^1.0.1
Unpacking  [####################################]  100%
Library Manager: UrlEncode@1.0.1 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
fatal: no és un repositori de git (ni cap pare fins al punt de muntatge /)
S'atura a la frontera de sistema de fitxers (GIT_DISCOVERY_ACROSS_FILESYSTEM no està establert).
Version: Unknown
Flag: -D REPO_VERSION=\"Unknown\"
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-8048S043C-smartdisplay.html
PLATFORM: Espressif 32 (6.5.0) > esp32-8048S043C
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32_smartdisplay @ 2.0.5+sha.4e689b5
|-- ArduinoJson @ 7.0.3
|-- UrlEncode @ 1.0.1
|-- lvgl @ 8.3.11
|-- Preferences @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
|-- Update @ 2.0.0
Building in release mode
Compiling .pio/build/esp32-8048S043C-smartdisplay/src/conf/global_config.cpp.o
Compiling .pio/build/esp32-8048S043C-smartdisplay/src/core/data_setup.cpp.o
Compiling .pio/build/esp32-8048S043C-smartdisplay/src/core/device/ESP32-2432S028R.cpp.o
Compiling .pio/build/esp32-8048S043C-smartdisplay/src/core/device/ESP32-3248S035C.cpp.o
Compiling .pio/build/esp32-8048S043C-smartdisplay/src/core/device/ESP32-smartdisplay.cpp.o
Compiling .pio/build/esp32-8048S043C-smartdisplay/src/core/files_query.cpp.o
[... 320 lines cut ...]
Compiling .pio/build/esp32-8048S043C-smartdisplay/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32-8048S043C-smartdisplay/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32-8048S043C-smartdisplay/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32-8048S043C-smartdisplay/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32-8048S043C-smartdisplay/libFrameworkArduino.a
Indexing .pio/build/esp32-8048S043C-smartdisplay/libFrameworkArduino.a
Linking .pio/build/esp32-8048S043C-smartdisplay/firmware.elf
Retrieving maximum program size .pio/build/esp32-8048S043C-smartdisplay/firmware.elf
Checking size .pio/build/esp32-8048S043C-smartdisplay/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  25.2% (used 82688 bytes from 327680 bytes)
Flash: [==        ]  17.3% (used 1132301 bytes from 6553600 bytes)
Building .pio/build/esp32-8048S043C-smartdisplay/firmware.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
=================================================================================================================== [SUCCESS] Took 149.74 seconds ===================================================================================================================

Environment                   Status    Duration
----------------------------  --------  ------------
esp32-8048S043C-smartdisplay  SUCCESS   00:02:29.744
==================================================================================================================== 1 succeeded in 00:02:29.744 ====================================================================================================================
 *  Terminal will be reused by tasks, press any key to close it. 
 ```

suchmememanyskill, 2024-02-28:
Might be an issue with the smartdisplay driver or my implementation then. Can you try to compile https://github.com/rzeldent/esp32-smartdisplay-demo ?

You may have to download this repository via the commandline: `git clone --recursive https://github.com/rzeldent/esp32-smartdisplay-demo`

themacboy, 2024-02-28:
Im tryed, but getting compile errors:

https://github.com/rzeldent/esp32-smartdisplay-demo/issues/25

themacboy, 2024-02-29:
After some learning and dificulties I build the demo, but get same problem:

 White screen 
 
 No error on monitor.

themacboy, 2024-03-02:
Hello ! Good and bad news lol

After a lot of reading, I found a thread that seems to solve the problem:

https://github.com/rzeldent/esp32-smartdisplay-demo/issues/21

For now I have edited the file esp32-8048S050C.json:

ST7262_PANEL_CONFIG_TIMINS_PCLK_HZ=(12.5 *1000000)

And now I can see the Wi-Fi screen selector (it's cool ;).

But the screen flips to portrait and the touch screen doesn't work.

Another question, is there a reason why you don't use rzeldent the original source? This seems more up to date.

themacboy, 2024-03-02:
I'm not really sure what I'm doing (I'm not good at C, lgvl and other coding hehehe), but I'll lay out my tests.

1. changes on platformio.ini
```
env]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = 
	#https://github.com/suchmememanyskill/esp32-smartdisplay
	https://github.com/rzeldent/esp32-smartdisplay
	bblanchon/ArduinoJson@^7.0.0
	plageoj/UrlEncode@^1.0.1
monitor_filters = esp32_exception_decoder
build_flags = 
	-DLV_CONF_PATH="../../../../src/conf/lv_conf.h"
extra_scripts = 
	pre:extract_commit.py
```

Result: Worked the same. Image is portraid and touch screen is not working.

themacboy, 2024-03-02:
2. Changes to esp32-8048S050C.json file:

I used the one from the thread referenced above and put in your custom variables/parameters (not sure what it's called in C)

```
{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "'-D ARDUINO_ESP32S3_DEV'",
      "'-D BOARD_HAS_PSRAM'",
      "'-D ARDUINO_USB_MODE=1'",
      "'-D ARDUINO_RUNNING_CORE=1'",
      "'-D ARDUINO_EVENT_RUNNING_CORE=1'",
      "'-D ARDUINO_USB_CDC_ON_BOOT=0'",
      "'-D ESP32_8048S050C'",
      "'-D LCD_WIDTH=800'",
      "'-D LCD_HEIGHT=480'",
      "'-D LVGL_BUFFER_PIXELS=(LCD_WIDTH*LCD_HEIGHT)'",
      "'-D LVGL_BUFFER_MALLOC_FLAGS=(MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT)'",
      "'-D GPIO_BCKL=2'",
      "'-D LCD_ST7262_PAR'",
      "'-D ST7262_PANEL_CONFIG_CLK_SRC=LCD_CLK_SRC_PLL160M'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=(12.5 *1000000)'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_H_RES=LCD_WIDTH'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_V_RES=LCD_HEIGHT'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_PULSE_WIDTH=4'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_BACK_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_FRONT_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_PULSE_WIDTH=4'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_BACK_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_FRONT_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_HSYNC_IDLE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_VSYNC_IDLE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_DE_IDLE_HIGH=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_ACTIVE_NEG=true'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_IDLE_HIGH=false'",
      "'-D ST7262_PANEL_CONFIG_DATA_WIDTH=16'",
      "'-D ST7262_PANEL_CONFIG_SRAM_TRANS_ALIGN=4'",
      "'-D ST7262_PANEL_CONFIG_PSRAM_TRANS_ALIGN=64'",
      "'-D ST7262_PANEL_CONFIG_HSYNC_GPIO_NUM=39'",
      "'-D ST7262_PANEL_CONFIG_VSYNC_GPIO_NUM=41'",
      "'-D ST7262_PANEL_CONFIG_DE_GPIO_NUM=40'",
      "'-D ST7262_PANEL_CONFIG_PCLK_GPIO_NUM=42'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R0=8'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R1=3'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R2=46'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R3=9'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R4=1'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G0=5'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G1=6'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G2=7'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G3=15'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G4=16'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G5=4'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B0=45'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B1=48'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B2=47'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B3=12'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B4=14'",
      "'-D ST7262_PANEL_CONFIG_DISP_GPIO_NUM=GPIO_NUM_NC'",
      "'-D ST7262_PANEL_CONFIG_FLAGS_DISP_ACTIVE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_FLAGS_RELAX_ON_IDLE=false'",
      "'-D ST7262_PANEL_CONFIG_FLAGS_FB_IN_PSRAM=true'",
      "'-D BOARD_HAS_TOUCH'",
      "'-D TOUCH_GT911_I2C'",
      "'-D GT911_I2C_HOST=I2C_NUM_0'",
      "'-D GT911_I2C_CONFIG_SDA_IO_NUM=19'",
      "'-D GT911_I2C_CONFIG_SCL_IO_NUM=20'",
      "'-D GT911_I2C_CONFIG_SDA_PULLUP_EN=GPIO_PULLUP_DISABLE'",
      "'-D GT911_I2C_CONFIG_SCL_PULLUP_EN=GPIO_PULLUP_DISABLE'",
      "'-D GT911_I2C_CONFIG_MASTER_CLK_SPEED=400000'",
      "'-D GT911_I2C_CONFIG_CLK_FLAGS=I2C_SCLK_SRC_FLAG_FOR_NOMAL'",
      "'-D GT911_IO_I2C_CONFIG_DEV_ADDR=ESP_LCD_TOUCH_IO_I2C_GT911_ADDRESS'",
      "'-D GT911_IO_I2C_CONFIG_CONTROL_PHASE_BYTES=1'",
      "'-D GT911_IO_I2C_CONFIG_DC_BIT_OFFSET=0'",
      "'-D GT911_IO_I2C_CONFIG_LCD_CMD_BITS=16'",
      "'-D GT911_IO_I2C_CONFIG_LCD_PARAM_BITS=0'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DC_LOW_ON_DATA=false'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DISABLE_CONTROL_PHASE=true'",
      "'-D GT911_TOUCH_CONFIG_X_MAX=LCD_WIDTH'",
      "'-D GT911_TOUCH_CONFIG_Y_MAX=LCD_HEIGHT'",
      "'-D GT911_TOUCH_CONFIG_RST_GPIO_NUM=38'",
      "'-D GT911_TOUCH_CONFIG_INT_GPIO_NUM=18'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_RESET=0'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_INTERRUPT=0'",
      "'-D TOUCH_SWAP_XY=false'",
      "'-D TOUCH_SWAP_X=false'",
      "'-D TOUCH_SWAP_Y=false'",
      "'-D BOARD_HAS_TF'",
      "'-D TF_CS=10'",
      "'-D TF_SPI_MOSI=11'",
      "'-D TF_SPI_SCLK=12'",
      "'-D TF_SPI_MISO=13'",

      "-DCYD_SCREEN_GAP_PX=10",
      "-DCYD_SCREEN_MIN_BUTTON_HEIGHT_PX=40",
      "-DCYD_SCREEN_MIN_BUTTON_WIDTH_PX=40",
      "-DCYD_SCREEN_FONT=lv_font_montserrat_16",
      "-DCYD_SCREEN_FONT_SMALL=lv_font_montserrat_12",
      "-DCYD_SCREEN_SIDEBAR_SIZE_PX=50",
      "-DCYD_SCREEN_DRIVER_ESP32_SMARTDISPLAY=1",
      "-DCYD_SCREEN_DISABLE_TOUCH_CALIBRATION=1",
      "-DCYD_SCREEN_DISABLE_INVERT_COLORS=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "esp32-8048S050C",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.aliexpress.com/item/1005005938915207.html",
  "vendor": "Sunton"
}
```

RESULT: Worked the same. Image is portraid and touch screen is not working.

themacboy, 2024-03-02:
Trying to figure how to console log the touch inputs, to be sure that touch screen is activated or not. In the previous demo you posted it works great.

GAZ082, 2024-03-02:
I'll be watching this. Same display.

themacboy, 2024-03-02:
ummm Noticed that touch screen is really working.

After screen turns to sleep mode, If i touch the screen then it wake up.

Then not sure why touch screen is not working to select a network.

suchmememanyskill, 2024-03-02:
I just re-read the config. All other CYD screens are vertical screens, but this one is actually horizontal. That means i have misconfigured it. Will push out an update later today with your findings too to correct for this

suchmememanyskill, 2024-03-02:
@themacboy just pushed a new build to the 4.3inch branch. Please try to compile and run via platformio

themacboy, 2024-03-02:
Working great now! I review all the code you made! Really fast!

Your effort is greatly appreciated!!!

Now I'm trying some visual tweaks like a sidebar and bigger icons and a bigger font size.

In the meantime I'm doing tests and I'll report the results.

Thank you so much!

suchmememanyskill, 2024-03-02:
@themacboy Sick!

Can you post a screenshot of the screen so i can try to make those ajustments for you?

If you want to try yourself, these are the values you need to edit in the boards/.json file
```
        "-DCYD_SCREEN_GAP_PX=10",
        "-DCYD_SCREEN_MIN_BUTTON_HEIGHT_PX=40",
        "-DCYD_SCREEN_MIN_BUTTON_WIDTH_PX=40",
        "-DCYD_SCREEN_FONT=lv_font_montserrat_16",
        "-DCYD_SCREEN_FONT_SMALL=lv_font_montserrat_12",
        "-DCYD_SCREEN_SIDEBAR_SIZE_PX=50",
```

To support different font sizes, enable them in src/conf/lv_conf.h

```c
/*==================
 *   FONT USAGE
 *===================*/

/*Montserrat fonts with ASCII range and some symbols using bpp = 4
 *https://fonts.google.com/specimen/Montserrat*/
#define LV_FONT_MONTSERRAT_8  0
#define LV_FONT_MONTSERRAT_10 1
#define LV_FONT_MONTSERRAT_12 1
#define LV_FONT_MONTSERRAT_14 1
#define LV_FONT_MONTSERRAT_16 1
#define LV_FONT_MONTSERRAT_18 0
#define LV_FONT_MONTSERRAT_20 0
#define LV_FONT_MONTSERRAT_22 0
#define LV_FONT_MONTSERRAT_24 0
#define LV_FONT_MONTSERRAT_26 0
#define LV_FONT_MONTSERRAT_28 1
#define LV_FONT_MONTSERRAT_30 0
#define LV_FONT_MONTSERRAT_32 0
#define LV_FONT_MONTSERRAT_34 0
#define LV_FONT_MONTSERRAT_36 0
#define LV_FONT_MONTSERRAT_38 0
#define LV_FONT_MONTSERRAT_40 0
#define LV_FONT_MONTSERRAT_42 0
#define LV_FONT_MONTSERRAT_44 0
#define LV_FONT_MONTSERRAT_46 0
#define LV_FONT_MONTSERRAT_48 0
```

If you found a nice combination, please show a screenshot and the accompanying configuration

themacboy, 2024-03-02:
Oh ! Yes I noticed it before and just make a PR with changes.

closed 2024-03-02 by themacboy

suchmememanyskill, 2024-03-02:
Thank you! Please do make sure to make a pull request to the 4.3inch branch, so i can merge it into dev when ready

themacboy, 2024-03-02:
@GAZ082 check the last PR

themacboy, 2024-03-03:
Btw I made a case for the ESP32-8048S043C:

[image]

[image]

[image]

[image]

themacboy, 2024-03-03:
https://github.com/suchmememanyskill/CYD-Klipper/assets/8063711/4dc38c7b-0210-4dcb-9e62-fdf0e0bd5054

themacboy, 2024-03-05:
@suchmememanyskill love this project:

[image]