suchmememanyskill/CYD-Klipper #59: Layer progress not shown

szafranski, 2024-03-07:
[image]

As shown above

suchmememanyskill, 2024-03-07:
This is an issue with your configuration

https://moonraker.readthedocs.io/en/latest/printer_objects/#print_stats see 'info'

szafranski, 2024-03-07:
Thank you, but looks like virtual_sdcard has to be addod so it would be active. I have this feature so I am not sure what I am doing wrong.

suchmememanyskill, 2024-03-07:
I don't know what you are looking at, but it's not what i sent you

[image]

szafranski, 2024-03-07:
Oh, my bad. Thank you so much for your time!

closed 2024-03-07 by szafranski