suchmememanyskill/CYD-Klipper #60: Blank screen on 3248S035C

feletoch, 2024-03-08:
`
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Hello World
[    36][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: global_config NOT_FOUND
Config version: 0
Clearing Global Config
[    48][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: global_config NOT_FOUND
[  1167][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  2285][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  3403][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  4521][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  5639][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  6757][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  7875][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[  8993][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 10111][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 11229][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 12347][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 13465][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 14583][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 15701][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 16819][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 17937][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 19055][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 20173][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 21291][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 22409][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 23527][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 24645][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 25763][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 26881][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 27999][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 29117][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 30235][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 31353][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 32471][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 33589][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 34707][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 35825][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 36943][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 38061][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 39179][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 40297][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 41415][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 42533][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 43651][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 44769][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 45887][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 47005][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 48123][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 49241][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 50359][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 51477][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 52595][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 53713][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 54831][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 55949][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 57067][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 58185][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 59303][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 60421][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 61539][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 62657][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[... 32 lines cut ...]
[ 99551][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[100669][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[101787][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[102905][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[104023][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[105141][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[106259][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[107377][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[108495][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[109613][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[110731][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[111849][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[112967][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[114085][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core  1 register dump:
PC      : 0x401969ab  PS      : 0x00060c36  A0      : 0x801911ad  A1      : 0x3ffb0330  
A2      : 0x3ffb81e8  A3      : 0x3ffb0670  A4      : 0x3ffb0554  A5      : 0x00001000  
A6      : 0x00000006  A7      : 0x3ffb0594  A8      : 0xa0000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x10000000  A12     : 0x00000000  A13     : 0x00060223  
A14     : 0x00060223  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008af29  LEND    : 0x4008af39  LCOUNT  : 0xffffffff  

Backtrace: 0x401969a8:0x3ffb0330 0x401911aa:0x3ffb0360 0x4019486a:0x3ffb0670 0x401948a6:0x3ffb0700 0x40108fd5:0x3ffb0740 0x40136cb0:0x3ffb0780 0x40187a6d:0x3ffb0800 0x40187b7d:0x3ffb0840 0x40101194:0x3ffb0860 0x4010122a:0x3ffb0880 0x4010119b:0x3ffb08a0 0x4010122a:0x3ffb08c0 0x4010119b:0x3ffb08e0 0x4010122a:0x3ffb0900 0x4010119b:0x3ffb0920 0x4010122a:0x3ffb0940 0x4010119b:0x3ffb0960 0x4010122a:0x3ffb0980 0x4010119b:0x3ffb09a0 0x4010122a:0x3ffb09c0 0x4010119b:0x3ffb09e0 0x4010122a:0x3ffb0a00 0x4010119b:0x3ffb0a20 0x4010122a:0x3ffb0a40 0x4010119b:0x3ffb0a60 0x4010122a:0x3ffb0a80 0x4010119b:0x3ffb0aa0 0x4010122a:0x3ffb0ac0 0x4010119b:0x3ffb0ae0 0x4010122a:0x3ffb0b00 0x4010119b:0x3ffb0b20 0x4010122a:0x3ffb0b40 0x4010119b:0x3ffb0b60 0x4010122a:0x3ffb0b80 0x4010119b:0x3ffb0ba0 0x4010122a:0x3ffb0bc0 0x4010119b:0x3ffb0be0 0x4010122a:0x3ffb0c00 0x4010119b:0x3ffb0c20 0x4010122a:0x3ffb0c40 0x4010119b:0x3ffb0c60 0x4010122a:0x3ffb0c80 0x4010119b:0x3ffb0ca0 0x4010122a:0x3ffb0cc0 0x4010119b:0x3ffb0ce0 0x4010122a:0x3ffb0d00 0x4010119b:0x3ffb0d20 0x4010122a:0x3ffb0d40 0x4010119b:0x3ffb0d60 0x4010122a:0x3ffb0d80 0x4010119b:0x3ffb0da0 0x4010122a:0x3ffb0dc0 0x4010119b:0x3ffb0de0 0x4010122a:0x3ffb0e00 0x4010119b:0x3ffb0e20 0x4010122a:0x3ffb0e40 0x4010119b:0x3ffb0e60 0x4010122a:0x3ffb0e80 0x4010119b:0x3ffb0ea0 0x4010122a:0x3ffb0ec0 0x4010119b:0x3ffb0ee0 0x4010122a:0x3ffb0f00 0x4010119b:0x3ffb0f20 0x4010122a:0x3ffb0f40 0x4010119b:0x3ffb0f60 0x4010122a:0x3ffb0f80 0x4010119b:0x3ffb0fa0 0x4010122a:0x3ffb0fc0 0x4010119b:0x3ffb0fe0 0x4010122a:0x3ffb1000 0x4010119b:0x3ffb1020 0x4010122a:0x3ffb1040 0x4010119b:0x3ffb1060 0x4010122a:0x3ffb1080 0x4010119b:0x3ffb10a0 0x4010122a:0x3ffb10c0 0x4010119b:0x3ffb10e0 0x4010122a:0x3ffb1100 0x4010119b:0x3ffb1120 0x4010122a:0x3ffb1140 0x4010119b:0x3ffb1160 0x4010122a:0x3ffb1180 0x4010119b:0x3ffb11a0 0x4010122a:0x3ffb11c0 0x4010119b:0x3ffb11e0 0x4010122a:0x3ffb1200 0x4010119b:0x3ffb1220 0x4010122a:0x3ffb1240 0x4010119b:0x3ffb1260 0x4010122a:0x3ffb1280 0x4010119b:0x3ffb12a0 0x4010122a:0x3ffb12c0 0x4010119b:0x3ffb12e0 0x4010122a:0x3ffb1300 0x4010119b:0x3ffb1320 0x4010122a:0x3ffb1340 0x4010119b:0x3ffb1360 0x4010122a:0x3ffb1380 0x4010119b:0x3ffb13a0 0x4010122a:0x3ffb13c0 0x4010119b:0x3ffb13e0 |<-CONTINUES`

suchmememanyskill, 2024-03-08:
Is the screen defective?

The first time you plug it in it should be running a demo of some sort. Did you see this demo?

Error 263 i believe means timeout

feletoch, 2024-03-08:
yep, there was a demo, but I don't remember if i could interact with it.

suchmememanyskill, 2024-03-09:
I have no idea why this is happening

Could you open the project in visual studio code + platformio, and upload the smartdisplay variant of the screen to your device and see if that works?

feletoch, 2024-03-09:
Tried both the master branch and dev branch. managed to get my other 2432S032C working, which is bonus.

To test the panel, I flash the firmware of another project (openHASP) and confirm the panel is working fine.

suchmememanyskill, 2024-03-09:
Can you link that project so i can cross reference their driver and my driver?

feletoch, 2024-03-09:
https://github.com/HASwitchPlate/openHASP

Ali-ElCheikh, 2024-04-16:
Did you end up figuring out what the issue is was? I am having the same issue on the resistive 3.5" screen

feletoch, 2024-04-16:
Didn't have time to try the method above, maybe you can have a go and let me know

Ali-ElCheikh, 2024-04-16:
Is there any output on your screen? My screen seems to not output anything at all after the install - it was outputting the original "testing" material when I first got it and turned it on. I tried an i2C scanner and the thing above and nothing happened - only LED works

suchmememanyskill, 2024-04-16:
Small curiosity i have; the stack trace displayed earlier is incomplete.

(Specifically, it is crashing at init of the display or init of the touch?)

https://github.com/suchmememanyskill/CYD-Klipper/blob/dev/Manual_Install.md#building--flashing-via-platformio 

Please select the 'esp32-3248S035C' device from the list. This should give additional info on a crash.

suchmememanyskill, 2024-06-28:
Issue is stale, closing

closed 2024-06-28 by suchmememanyskill

Gauner101184, 2025-02-19:
Hi,

i have the exact same problem with my brand new 3248S035C display from Aliexpress.
The Demo worked fine, but after flashing CYD Klipper, the exact same error occours.

Gauner101184, 2025-02-19:
This is what happens after flashing via Visual Code...
The display is flashing and reboots all time.

Writing at 0x0010de5e... (80 %)
Writing at 0x001139ae... (82 %)
Writing at 0x001190cb... (84 %)
Writing at 0x0012173e... (86 %)
Writing at 0x00128f70... (88 %)
Writing at 0x0012fc93... (90 %)
Writing at 0x00134cd2... (92 %)
Writing at 0x0013a5ff... (94 %)
Writing at 0x0013fdd1... (96 %)
Writing at 0x00145037... (98 %)
Writing at 0x0014aefe... (100 %)
Wrote 1291568 bytes (836693 compressed) at 0x00010000 in 20.3 seconds (effective 508.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on COM9 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
CYD-Klipper Unknown
Type 'help' for serial console command list
> [    27][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: global_config NOT_FOUND
[    31][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: global_config NOT_FOUND
E (251) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (272) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (272) GT911: touch_gt911_read_cfg(354): GT911 read error!
E (273) GT911: esp_lcd_touch_new_i2c_gt911(124): GT911 init failed
E (279) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4008deb0
file: ".pio/libdeps/esp32-3248S035C-SD/esp32_smartdisplay/src/lvgl_gt911_i2c.c" line 252
func: lvgl_touch_init
expression: esp_lcd_touch_new_i2c_gt911(io_handle, &touch_config, &touch_handle)

abort() was called at PC 0x4008deb3 on core 1

Backtrace: 0x40084009:0x3ffd6780 0x4008debd:0x3ffd67a0 0x40093df1:0x3ffd67c0 0x4008deb3:0x3ffd6840 0x4010a154:0x3ffd6860 0x40109d78:0x3ffd68f0 0x400d5b1b:0x3ffd6910 0x400df085:0x3ffd6930 0x401179fe:0x3ffd6960      

  #0  0x40084009:0x3ffd6780 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008debd:0x3ffd67a0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40093df1:0x3ffd67c0 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x4008deb3:0x3ffd6840 in _esp_error_check_failed at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_err.c:43
  #4  0x4010a154:0x3ffd6860 in lvgl_touch_init at .pio/libdeps/esp32-3248S035C-SD/esp32_smartdisplay/src/lvgl_gt911_i2c.c:183 (discriminator 1)
  #5  0x40109d78:0x3ffd68f0 in smartdisplay_init at .pio/libdeps/esp32-3248S035C-SD/esp32_smartdisplay/src/esp32_smartdisplay.c:267
  #6  0x400d5b1b:0x3ffd6910 in screen_setup() at src/core/device/ESP32-smartdisplay.cpp:55
  #7  0x400df085:0x3ffd6930 in setup() at src/main.cpp:18
  #8  0x401179fe:0x3ffd6960 in loopTask(void*) at C:/Users/gaune/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 38cafa5b0f77659f

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
[... 13 lines cut ...]
E (273) GT911: esp_lcd_touch_new_i2c_gt911(124): GT911 init failed
E (279) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4008deb0
file: ".pio/libdeps/esp32-3248S035C-SD/esp32_smartdisplay/src/lvgl_gt911_i2c.c" line 252
func: lvgl_touch_init
expression: esp_lcd_touch_new_i2c_gt911(io_handle, &touch_config, &touch_handle)

abort() was called at PC 0x4008deb3 on core 1

Backtrace: 0x40084009:0x3ffd6780 0x4008debd:0x3ffd67a0 0x40093df1:0x3ffd67c0 0x4008deb3:0x3ffd6840 0x4010a154:0x3ffd6860 0x40109d78:0x3ffd68f0 0x400d5b1b:0x3ffd6910 0x400df085:0x3ffd6930 0x401179fe:0x3ffd6960      

  #0  0x40084009:0x3ffd6780 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008debd:0x3ffd67a0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40093df1:0x3ffd67c0 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x4008deb3:0x3ffd6840 in _esp_error_check_failed at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_err.c:43
  #4  0x4010a154:0x3ffd6860 in lvgl_touch_init at .pio/libdeps/esp32-3248S035C-SD/esp32_smartdisplay/src/lvgl_gt911_i2c.c:183 (discriminator 1)
  #5  0x40109d78:0x3ffd68f0 in smartdisplay_init at .pio/libdeps/esp32-3248S035C-SD/esp32_smartdisplay/src/esp32_smartdisplay.c:267
  #6  0x400d5b1b:0x3ffd6910 in screen_setup() at src/core/device/ESP32-smartdisplay.cpp:55
  #7  0x400df085:0x3ffd6930 in setup() at src/main.cpp:18
  #8  0x401179fe:0x3ffd6960 in loopTask(void*) at C:/Users/gaune/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 38cafa5b0f77659f

Rebooting...
ets Jul 29 2019 12:21:46