suchmememanyskill/CYD-Klipper #61: :ENHANCEMENT: Request for board ESP32-2432S024.

Shalasere, 2024-03-09:
Cheers! I'm quite impressed with what's been done. I'm wondering if it would be possible to add the following board: ESP32-2432S024.

Checking into things, the model is listed in [ESP32-Cheap-Yellow-Display](https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Variants) and [esp32-smartdisplay](https://github.com/rzeldent/platformio-espressif32-sunton/blob/e75f05f68229c59de4c6aa40e38a16e012e5883f/esp32-2432S024C.json)

The reason I believe this would be a worthwhile inclusion is that it's the standard 2.4 in size, just with the type-C port and is a capacitive touch display. My thought is that the implementation might not be so painful given you've already include a different capacitive touchscreen in the form of the larger screen size. 

I'm happy to attempt any testing and/or fetch any required further documentation to help the implementation along.

suchmememanyskill, 2024-03-09:
@Shalasere would you mind testing out the following build?

Actions build (use to download merged firmware): https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8216965803
Branch (use to download code): https://github.com/suchmememanyskill/CYD-Klipper/tree/dev
Build/Run instructions: https://github.com/suchmememanyskill/CYD-Klipper/blob/experimental-multi-printer/Build.md

Shalasere, 2024-03-09:
Doing so now. 
Just to note, The build/run calls out a merged-firmware.bin, while the firmware from the actions is merged_firmware.bin.

Shalasere, 2024-03-09:
After flashing the merged_firmware.bin, Screen popped right up searching for a network, capacitive touch even kind of works, though seems it might need calibration.

Shalasere, 2024-03-09:
I had slight trouble following this set of instructions:

> Expand the folder/tab for your specific screen
> Entries with the suffix '-SD' are using the smartdisplay driver. Entries without this suffix are using a custom driver
> Usually, a custom driver is preferred over the smartdisplay driver

As I was only seeing the explorer window on the left and believed that I needed to be doing something with my specific board.json file. It may be valuable to include something like "Please verify that upon clicking on the Alien/Bug Icon, you should see the "Project Tasks" option on the left and then may select the board-specific project from there."

Further, upon launching "Upload and Monitor" , I received the following error:
`Processing esp32-2432S024C-SD (board: esp32-2432S024C-smartdisplay; platform: espressif32; framework: arduino)
-------------------------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/suchmememanyskill/esp32-smartdisplay
UserSideException: Please install Git client from https://git-scm.com/downloads`

Not a huge issue as most folks will have it installed, but something that might be a good heads-up to include as a req.
Simply installing through [choco](https://chocolatey.org) relaunching vscode fixed it. 

I went ahead and flashed it and I've attached my compilation log. 
[esp32-2432S024C-SD_compilation.log](https://github.com/suchmememanyskill/CYD-Klipper/files/14547910/esp32-2432S024C-SD_compilation.log)

Shalasere, 2024-03-09:
Between the 2 units I have carried this out on, the touch calibration is completely off, but thankfully off in a similar way, taking the preloaded demo, I validated that the entirety of the screen has touch functionality, yet on these models only the center of screen seems to react and further, I believe the touch inputs are flipped and inverted "r is b, f is v, d is spacebar, y is j". This is the same between both units, however, there is the occasional vertical white line that pops up on one unit, and the other is mostly vertical jiggling white lines from the first line of "Enter Wifi Password" prompt down.

This is the behavior between when I build it or when I simply flash the merged_firmware.bin

suchmememanyskill, 2024-03-09:
@Shalasere I indeed seemingly made a mistake.

New build: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8217753878

> Just to note, The build/run calls out a merged-firmware.bin, while the firmware from the actions is merged_firmware.bin.
I seemingly only changed this in the `experimental-multi-printer` branch. Not a huge issue, that'll likely get merged soon into dev. Thanks for telling me

Shalasere, 2024-03-10:
I've gone ahead and flashed the merged_firmware. The touch seems performing as expected! However, I'm still getting semi-inconsistent screen behavior with the white lines that I'd mentioned before. Attached below is a photo. This is happening similarly between 2 separate and brand-new devices. 
[image]

suchmememanyskill, 2024-03-10:
I have no idea what's going on here. Has it always done this or only since you flashed this software? Does it also do it here: https://github.com/rzeldent/esp32-smartdisplay-demo ?

Shalasere, 2024-03-10:
It was only after the flashing of the merged_firmware. I'll go about flashing that real quick and provide an update.

suchmememanyskill, 2024-03-11:
Whack, there's not a lot to configure with smartdisplay so i'm unsure what could be going wrong here.

Although, i am using v2.0.5 instead of latest (v2.0.7)

Shalasere, 2024-03-11:
I've been wracking my head on it a bit before bed. Comparing your .json to the one located on the demo-main, The only differences I was able to see was the renaming of a var on line 18, `GPIO_BCKL -> BCKL` and the var located on line 23 `"'-D ILI9341_SPI_BUS_MAX_TRANSFER_SZ=0->(LVGL_BUFFER_PIXELS*sizeof(lv_color16_t))'"` and that doesn't really matter in this situation as the lv_color_16_t located in the lvgl library is 0 anyway. (and has to do with some color translation, something I'm having no issues with. ) The rest of the implementations of this screen I found online aren't through smartdisplay, so that's not much help.

Shalasere, 2024-03-11:
Do you have suggestions on how I might attempt to use v2.0.7 instead of v2.0.5?

suchmememanyskill, 2024-03-11:
> the var located on line 23 "'-D ILI9341_SPI_BUS_MAX_TRANSFER_SZ=0->(LVGL_BUFFER_PIXELS*sizeof(lv_color16_t))'" and that doesn't really matter in this situation as the lv_color_16_t located in the lvgl library is 0 anyway.

Actually, could the lines be caused by trying to push too much data at once? Also lv_color_16_t probably isn't 0

For how to use it, it isn't super easy. I'll update it later tonight so you can test

suchmememanyskill, 2024-03-11:
Maybe you can try to downgrade the demo to v2.0.5 to see if you have the same issue again?

Shalasere, 2024-03-11:
I was thinking something similar, and began playing with the `"'-D ILI9341_SPI_CONFIG_PCLK_HZ=24000000'",` value. at 10MHZ same issue, at 0 it obviously complained loudly. I found an amazon listing that displayed a test project being uploaded at 80MHZ, but at that speed, the screen was distorted. I just tested it at 5MHZ and we're seeing the same affect. And that's a great idea, let me try that out.

suchmememanyskill, 2024-03-11:
Just to make sure also, you are not using something like the invert display setting in cyd klipper? In the smartdisplay driver that's something i wrote and may be broken

Shalasere, 2024-03-11:
Totally fair check. I'd seen that and stayed away from it, trying to limit what's goin' on. I've not set anything specifically, apart from slightly modifying the board.json in the ways I called out above.

Shalasere, 2024-03-11:
No luck on trying to launch the 2.0.5 using the 2.0.5 board defs. 
`Compiling .pio\build\esp32-2432S024C\lib025\WiFi\WiFi.cpp.o`
`Compiling .pio\build\esp32-2432S024C\lib025\WiFi\WiFiAP.cpp.o`
`Compiling .pio\build\esp32-2432S024C\lib025\WiFi\WiFiClient.cpp.o`
`Compiling .pio\build\esp32-2432S024C\lib025\WiFi\WiFiGeneric.cpp.o`
`Compiling .pio\build\esp32-2432S024C\lib025\WiFi\WiFiMulti.cpp.o`
`*** [.pio\build\esp32-2432S024C\lib018\esp32_smartdisplay\esp_lcd_touch.c.o] Error 1`
`*** [.pio\build\esp32-2432S024C\lib018\esp32_smartdisplay\esp32_smartdisplay.c.o] Error 1`
`*** [.pio\build\esp32-2432S024C\lib018\esp32_smartdisplay\esp_touch_cst816s.c.o] Error 1`
`*** [.pio\build\esp32-2432S024C\lib018\esp32_smartdisplay\lvgl_touch_cst816s_i2c.c.o] Error 1`
`============================================================= [FAILED] Took 7.03 seconds =============================================================`

Looks like loading up the project automatically resolved and updated some dependencies, not sure if that's what broke it or not.

suchmememanyskill, 2024-03-11:
@Shalasere new build https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8239898114

Shalasere, 2024-03-12:
Hell yeah! I think that did it! I just flashed one unit and we're flawless. Doing so on the second unit now.

suchmememanyskill, 2024-03-12:
> @Shalasere new build https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8239898114

@aam335 and @themacboy

Could you test the build above on the ESP32-4827S043C and ESP32-8048S043C respectively?

I updated the smartdisplay driver to fix the issue above, just want to make sure no regression took place

aam335, 2024-03-12:
> Could you test the build above on the ESP32-4827S043C and ESP32-8048S043C respectively?

ESP32-4827S043C tested - ok

themacboy, 2024-03-12:
I just compiled the DEV branch and flashed it on an ESP32-8048S043C.

It seems to work fine.

Should I check something specific?

suchmememanyskill, 2024-03-12:
Just if it works, thank you!

suchmememanyskill, 2024-03-15:
Released

closed 2024-03-15 by suchmememanyskill

Shalasere, 2024-03-16:
I know this was just completed today and I appreciate the work that's been put in both to update the chrome-based flasher (super impressive and needs to be far more present) but also the action merged_firmwares, but I couldn't leave it alone that I'd had inconsistent performance between the different screens I have. 

On line 41 of esp32-2432S028RV2-smartdisplay.json, you've got the following vendor specific config for the ILI9341 LCD driver board.
        `"'-D ILI9341_DEV_CONFIG_VENDOR_CONFIG=\"(ili9341_vendor_config_t[]){{.init_cmds=(ili9341_lcd_init_cmd_t[]){{.cmd=0xCF,.data=(uint8_t[]){0x00,0xC1,0x30},.data_bytes=3},{.cmd=0xED,.data=(uint8_t[]){0x64,0x03,0x12,0x81},.data_bytes=4},{.cmd=0xE8,.data=(uint8_t[]){0x85,0x00,0x78},.data_bytes=3},{.cmd=0xCB,.data=(uint8_t[]){0x39,0x2C,0x00,0x34,0x02},.data_bytes=5},{.cmd=0xF7,.data=(uint8_t[]){0x20},.data_bytes=1},{.cmd=0xEA,.data=(uint8_t[]){0x00,0x00},.data_bytes=2},{.cmd=0xC0,.data=(uint8_t[]){0x10},.data_bytes=1},{.cmd=0xC1,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0xC5,.data=(uint8_t[]){0x30,0x30},.data_bytes=2,},{.cmd=0xC7,.data=(uint8_t[]){0xB7},.data_bytes=1},{.cmd=0x3A,.data=(uint8_t[]){0x55},.data_bytes=1},{.cmd=0x36,.data=(uint8_t[]){0x08},.data_bytes=1},{.cmd=0xB1,.data=(uint8_t[]){0x00,0x1A},.data_bytes=2},{.cmd=0xB6,.data=(uint8_t[]){0x08,0x82,0x27},.data_bytes=3},{.cmd=0xF2,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0x26,.data=(uint8_t[]){0x01},.data_bytes=1},{.cmd=0xE0,.data=(uint8_t[]){0x0F,0x2A,0x28,0x08,0x0E,0x08,0x54,0xA9,0x43,0x0A,0x0F,0x00,0x00,0x00,0x00},.data_bytes=15},{.cmd=0xE1,.data=(uint8_t[]){0x00,0x15,0x17,0x07,0x11,0x06,0x2B,0x56,0x3C,0x05,0x10,0x0F,0x3F,0x3F,0x0F},.data_bytes=15},{.cmd=0x2B,.data=(uint8_t[]){0x00,0x00,0x01,0x3F},.data_bytes=4},{.cmd=0x2A,.data=(uint8_t[]){0x00,0x00,0x00,0xEF},.data_bytes=4},{.cmd=0x21},{.cmd=0x11,.delay_ms=120},{.cmd=0x29,.delay_ms=1}},.init_cmds_size=23}}\"'",`

If I take this and apply it to the esp32-2432s024C file on line 41, it completely fixes the line issues and jittering that was observed. The parameter needs to be changed from "NULL" to the above. However, I know this is no longer using the Espressif official board definitions, so that's a bit unfortunate.
[esp32-2432S024C-smartdisplay.json](https://github.com/suchmememanyskill/CYD-Klipper/files/14622716/esp32-2432S024C-smartdisplay.json)

KasperDGit, 2024-03-16:
@Shalasere thanks. that worked for my esp32 2432s024 Capacative Touch Display.
[image]

suchmememanyskill, 2024-03-16:
@Shalasere you mean this fixes the lines issue and such on your displays?

Feel free to send a pull request with this, if it makes the displays better

KasperDGit, 2024-03-16:
I was having the same vertical line issue.
My (lazy) way arround was that I switched to Invert Colors in Screen Settings, and that white lines ware not visible.
Now display is fixed after changing the line 41, as @Shalasere mentioned.
Thank You @suchmememanyskill

suchmememanyskill, 2024-03-18:
@KasperDGit @Shalasere 

I've pushed your fix to the dev branch, please give it a go to make sure i implemented it correctly

Shalasere, 2024-03-18:
Thank you, I apologize I've not had the opportunity to go in and push the
change. In about 4 hours, I'll be  able to try it out.

KasperDGit, 2024-03-18:
[image]
Tested, Ok...
Thank You.

Shalasere, 2024-03-19:
Tested on my 3 screens. Works like a charm.