suchmememanyskill/CYD-Klipper #67: Web install failure ...

themacboy, 2024-03-17:
Good morning,

when trying to update from web service I get this error:

[image]

There is no problem for me, I will compile on platformio, but I think you must know ;)

KulibinBoris, 2024-04-19:
Good afternoon.
I have the same problem, I can’t install the firmware via the web interface for the ESP32-8048S043, on 2.8" everything works fine, but the size is too small for me, I can’t see well. Could you help me?

suchmememanyskill, 2024-04-19:
@KulibinBoris see https://github.com/suchmememanyskill/CYD-Klipper/blob/dev/Manual_Install.md#building--flashing-via-platformio for how to install via platformio. Might be worthwhile to install via platformio anyway as you can tweak the size of buttons and text in cyd klipper

dronecz, 2024-06-10:
@suchmememanyskill I can not find proper JSON files (I guess that they are automaticaly generated) to make PR, but you have to change `"chipFamily": "ESP32"` to `"chipFamily": "ESP32-S3"` and flasher should work.

suchmememanyskill, 2024-06-10:
@dronecz thank you for the suggestion/fix.

The board definitions are present in /CYD-Klipper/boards

dronecz, 2024-06-10:
I mean in web definition. File with this content:

 `{"name": "to esp32-8048S043C-SD", "funding_url": "https://ko-fi.com/suchmememanyskill", "new_install_prompt_erase": true, "builds": [{"chipFamily": "ESP32", "parts": [{"path": "out/esp32-8048S043C-SD/bootloader.bin", "offset": 4096}, {"path": "out/esp32-8048S043C-SD/partitions.bin", "offset": 32768}, {"path": "out/esp32-8048S043C-SD/boot_app0.bin", "offset": 57344}, {"path": "out/esp32-8048S043C-SD/firmware.bin", "offset": 65536}]}]}`

suchmememanyskill, 2024-06-28:
@dronecz Can you test if this works now

closed 2024-06-28 by suchmememanyskill

dronecz, 2024-06-28:
> @dronecz Can you test if this works now

Yes, I can confirm that flashing works on my board with ESP32-S3 module.

commit 4ac87c8f (2024-06-11): Set chip family to ESP32-S3 for specific models (fix #67)
commit e2c2a38b (2024-06-28): v1.6.4 (#113)

* Fix gcode previews with special chars not loading

* Add .gitignore file (#108)

* Bulletproof ci.py (#107)

* Implement file sorting (implement #89)

* Set chip family to ESP32-S3 for specific models (fix #67)

* Add files menu to params panel while printing (implement #80)

* Update ci.py (#110)

Typo fix for ESP32-S3 boards array name

---------
commit 41aa073a (2024-08-02): Serial console (#117)

* Update readme

* Update README.md (#95)

* v1.6.4 (#113)

* Fix gcode previews with special chars not loading

* Add .gitignore file (#108)

* Bulletproof ci.py (#107)

* Implement file sorting (implement #89)

* Set chip family to ESP32-S3 for specific models (fix #67)

* Add files menu to params panel while printing (implement #80)

* Update ci.py (#110)

Typo fix for ESP32-S3 boards array name

---------

* Added serial console

* backspace, minor tweaks

* - added temporary_config alongside global_config, to hold non-persistent configuration
- added macros LOG, LOG [... cut]