suchmememanyskill/CYD-Klipper #69: No way to power "other" printer when having 2 and both are powered off

Misterke, 2024-03-20:
I have 2 Klipper printers and CYD-Klipper does now allow me to switch them both on and off *IFFFFF* at least one of them is powered.  If both are powered off, I get a screen showing a home icon at the top, a "Connecting to Klipper (Try xxx)" label and then a "Reset" button (which seems very dangerous as you then are changing IP addresses and cannot get back - but there seems to be another issue and fix for that) and a "Power Devices" button, which however only allows me to power on the "active" printer.

If I want to use the other printer, I first need to turn on the "active" printer, wait until the CYD-Klipper UI reflects this and then I can go to the "Printer" tab and there select "Power" for the other printer, turn it on and then go back to the Printer tab and "Power" for the printer that I did not need, to turn it off again.

I think the Printer tab should be always available and maybe replace that "Connecting" screen so that one can control the power for all the configured printers.

suchmememanyskill, 2024-03-20:
I've been refactoring the code to allow more and more of the ux to be into the main panel-based ui, but it makes a pretty big assumption there is at least one printer to pull data from. It'll take a pretty big refactor to change this.

For now, i could add a power menu to the printer select menu, but that'll also require a decent refactor as it currently only checks printers once the menu is opened. Hm

Misterke, 2024-03-20:
The "home" menu shows both printers, one "Offline", the other "Active" - even though both are turned off (and then also the Klipper service for them is not running, but Moonraker is).  There being able to change the "Active" printer from one to the other (still both powered off) would already help, since then going back to the "Connecting to Klipper" screen would now be for the newly activated printer and I would assume also the "Power Devices" would then be for that printer.  Still not very user friendly, but usable ...

suchmememanyskill, 2024-03-20:
Could you try this dev build: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8366079312
(How to use developer builds: https://github.com/suchmememanyskill/CYD-Klipper/blob/master/Build.md)

Misterke, 2024-03-21:
I'll try it this evening when I'm near my printers again ;-)

Misterke, 2024-03-21:
I flashed the indicated version, but I did build it with platformio instead of taking the build you mentioned (I did use the exact same commit as basis though).

Let me start by saying that I immensely appreciate all you are doing and how you are addressing the issues users are raising.  The XTouch project for the Bambu printers shows that this is not evident and that a great project can quickly bleed dry if the developer loses interest.  So many Kudos for your efforts.

The development version is also an immens step forward and it does allow me to turn on the printer I need, but ... "everything can be improved", so just in that context I'd like to illustrate what I think is not so logical and could be improved.

1. I would put the "Power" control solely in the "Printer" view.  Now the active printer's power (irrespectively whether it is on or off) can be found in the "Macro" view, but the power control of the other printers are in the "Printer" view.  It would be way more logical to have it always in the "Printer" view (and have that available irrespectively of whether you have one or more printers).
2. I always found it weird that the "Screen Settings" are in the "Macro" view.  I have also noticed that several of these settings are specific to the active printer (i.e. the color of the screen changes when you change the printer you are controlling).  This is very nice as it can really help identify what printer you are on, but ... that also would make it more logical to move the settings to the "Printer" view rather than having it under "Macro".  That way the "Macro" view can also be removed as long as the controlled printer is not actually connected (so is off).  That would mean that each printer always keeps the "Power" button instead of having it become "Control" for printers that are on.
3. Switching control from one printer to another could just be done by clicking the top part of the printer (with name, status and progress) in that "Printer" view, not requiring a separate button.  It would also be better to allow that switching to printers that are turned off.  Since you can turn off the printer you are controlling, you can always end up with a printer in OFF state being under control, so then it also would be logical that you can switch control to printers that are already in OFF state.  Especially since some settings seem printer-specific, it makes sense that you could get to those without having to turn on each printer in turn. 
4. As long as no printer is "connected", I would keep the GUI on the "Printer" view - even if you have only configured a single printer.  There you could visualize whether the printer in control is still connecting or is connected, making the "Link" view no longer needed.

I realize that the above is pushing to make the "Printer" view very prominent, but I do think the selection of the printer is the logical top-level.  Also in KlipperScreen that is the case and in things like MobileRaker (Android app for controlling Klipper printers).

Then there are a couple of things I noticed, but which might be specific to this dev branch:
- The FPS and CPU counter could be nice, but if intended to be there in release versions, then they would be better get a dedicated spot where they don't cover other GUI items
- I regularly saw the device rebooting.  Now this might have been linked to a bad WIFI connection as I was testing this a bit farther away from my AP, but is shortly losing connection a reason for a full restart?  Below the serial prints for such cycles:

```
Failed to fetch printer data: 503
Failed to fetch printer data: 503
Failed to fetch printer data: 503
CPU Speed: 80 MHz
Failed to fetch printer data: 503
[ 67625][E][WiFiGeneric.cpp:1582] hostByName(): DNS Failed for octopi
Failed to fetch printer data: -1
[ 68859][E][WiFiSTA.cpp:305] begin(): config failed
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Hello World
Config version: 5
Screen init done
Connecting to Q2 with a password length of 63
OTA Update Result: -3
Failed to fetch printer data: 503
```

suchmememanyskill, 2024-03-21:
> Let me start by saying that I immensely appreciate all you are doing and how you are addressing the issues users are raising.

❤️

> I would put the "Power" control solely in the "Printer" view. Now the active printer's power (irrespectively whether it is on or off) can be found in the "Macro" view, but the power control of the [...]

I do disagree slightly with this, as CYD-Klipper is designed around having a singular printer connected. I can turn the current device's control button into a power control button, and keep the macros menu as-is.

> I always found it weird that the "Screen Settings" are in the "Macro" view. I have also noticed that several of these settings are specific to the active printer (i.e. the color of the screen [...]

Yeah, fair. For single printer use it to me makes sense as it's the only menu with space for such a settings button. It's not something you access often either. But i do agree for multi printer scenarios it's slightly odd to have the settings be in the macros screen. Proposing to turn the delete icon of the current printer into a settings icon.

> Switching control from one printer to another could just be done by clicking the top part of the printer (with name, status and progress) in that "Printer" view, not requiring a separate button. It [...]

I feel like this wouldn't be a very intuitive way to control a device. I can look into not disabling the control button on turned off devices to see how that works UX wise.

> As long as no printer is "connected", I would keep the GUI on the "Printer" view - even if you have only configured a single printer. There you could visualize whether the printer in control is still connecting or is connected, making the "Link" view no longer needed.

My past self is biting me in the ass here; the printer view updates roughly every 20 seconds. This isn't ideal for checking if a printer is online or not. The link screen checks every 1s. Also i'd prefer to only show the printer view if there's actually multiple printers.

> The FPS and CPU counter could be nice, but if intended to be there in release versions, then they would be better get a dedicated spot where they don't cover other GUI items

FPS/CPU usage is only shown on developer builds (builds that do not directly have a release tied to them). I don't want to show this on release builds

> I regularly saw the device rebooting. Now this might have been linked to a bad WIFI connection as I was testing this a bit farther away from my AP, but is shortly losing connection a reason for a full restart? Below the serial prints for such cycles:

There are some wifi related crashes i still have to investigate. Currently, CYD-Klipper makes no attempt to reconnect to wifi and reboots the device instead if it loses connection (i should probably add a log for this). Maybe in the future i can add some sort of retry logic

The dev branch in general is a bit unstable as i haven't tested everything thouroughly yet

Misterke, 2024-03-21:
@suchmememanyskill , just take my list as my suggestions.  Having the power control spread between printer and macros menu is the most annoying one to me.  Just having power always "also" present in the printer view for all printers irrespective of their power state would already address that.   But as said, even now this version at least allows me to turn the appropriate printer on without having to first turn the printer that was last controlled on, so it is already a major improvement.

Personally I think the Printer view also makes sense with just a single printer, since also with just one printer you basically have to add its IP/hostname and other parameters and have a way to change those and as said, it would also make sense to have settings accessible via that Printer menu as some of those are printer-specific even if you only have a single printer.  Anyway, GUI stuff is always "taste", so I'll let it rest here and leave it up to you to figure out the best and feasible next step.

Again, many thanks for your work.  If the current change gets merged, I guess the issue can be closed.

suchmememanyskill, 2024-03-22:
@Misterke please try the latest build. I have implemented the settings button in the printer menu, the power button on the active printer and retrying to connect to a lost wifi network. (I also implemented `["display_status"]["progress"]` from your other issue).

The only thing i did not implement is allowing to switch to an offline printer. Besides causing weird crashes, i really do not see the point of switching to an offline printer

Misterke, 2024-03-25:
> @Misterke please try the latest build. I have implemented the settings button in the printer menu, the power button on the active printer and retrying to connect to a lost wifi network.

Works perfectly!  I do see the reconnection happening regularly when ex. a printer comes online or goes offline, but it isn't really disturbing the flow too much.

> The only thing i did not implement is allowing to switch to an offline printer. Besides causing weird crashes, i really do not see the point of switching to an offline printer

I can live with that.

For the timing I suggest to discuss on the other issue.  I've also created a PR for getting the elapsed time correctly reflected (without messing up your estimations for the remaining time).

Leaves the question: shall I close this issue or should I wait until it's merged into a release?

closed 2024-03-29 by suchmememanyskill