suchmememanyskill/CYD-Klipper #70: Intermittent WiFi Connection Drops

Darkstar77, 2024-03-20:
I have 3 CYD's setup, the 2 original, and one larger 3.5 model. All of them suffer from intermittent WiFi connectivity issues. Basically they lose WiFi connectivity, and I get a "Reset Wifi" button. Press it, and are my AP's, which I can then connect to instantly. I'm not really sure what's going on, but I did observe the following:

1. It's random, both on which device will hit this, and the duration between. I would say on average it's 2 or 3 times a week across all 3 devices. Which device has the issue seems to be random as well. They all experience it.
2. When it happens, click the reset button, initially the AP is not listed and I need to refresh them. The AP is then visible right away.
3. I have a standalone AP with older A/B/G/N connectivity I use for smart home devices, and I have a mesh network with Wifi 5 and Wifi 6e nodes for all the end user devices in my home. My printers are on the mesh network. Which network I am connected to with the CYD does not impact this issue.

Based on my observations, and the second point above specifically, it's like the CYD is losing the AP for some reason, and just going into a failed state. Rescanning brings it right back, and it connects right away. I don't know how much of the WiFi connection logic is controlled by your code, but you might want to add a WiFi network refresh as part of the connection retry, since the AP is not visible when I click reset, but is immediately visible after clicking refresh. Performing this refresh before retrying the connection might be needed to make it more fault tolerant.

Baring that, could you please add a feature to save the AP passwords? If it just dropped now and again, and I could click on the AP to reconnect, that would honestly be great. Having to enter the dang password, and I use a longish one at 30 characters, is a real pain in the butt. It's annoying enough that I'm submitting this as an issue. Saving the password would make this 10x less annoying.

Thanks for all your hard work on this project, I think it's great. I've gone so far as to create several cases for it, with various mounting solutions: https://www.printables.com/model/748504-cyd-klipper-mounts

suchmememanyskill, 2024-03-20:
@Darkstar77 needing to re-enter the wifi credentials seems weird to me. What happens instead when you restart the device, if it gets stuck in an infinite loop trying to connect to the wifi network.

For how much i manage wifi on the device, very little. The arduino framework has a wifi begin call that starts connecting, more than that i do not do.

It may be possible that pinning core 0 with doing background requests frequently to klipper is causing this, as afaik core 0 is responsible for managing such hardware

suchmememanyskill, 2024-03-20:
@Darkstar77 could you try this dev build: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8366079312
(How to use developer builds: https://github.com/suchmememanyskill/CYD-Klipper/blob/master/Build.md)

It literally just restarts the screen once it found a password equal to your SSID. It sounds like a restart would fix the issue though; CYD-Klipper does save your wifi creds but rebooting then immediately trying to connect to the previously broken wifi connection may cause hangs

Darkstar77, 2024-03-21:
@suchmememanyskill I installed the build you asked me to on one of my devices. I'll let you know if I continue to see this issue. Should happen over the next ~3 days if it's going to happen.

My use case has my printers on all the time, with the CYD screens powered from the serial pins feeding +5V from the printer (just the +5v and the GND pins are connected). Works great, but makes restarting the screen a little difficult. I didn't want to power it via USB, since the cable sticks out quite a lot, and it's micro USB instead of USB type C. The result is a nice integration to my printers, but getting at the reset button, or power cycling just the CYD is not easy, and requires a minor disassembly. I'll try that on one of the devices not running the dev build next time they hit that state and see what it does.

suchmememanyskill, 2024-03-22:
@Darkstar77 i also just pushed a new build to the dev branch (https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8396463313) that attempts to reconnect to wifi once. It freezes the screen while it attempts to reconect (for about 15 seconds, if it takes longer it restarts)

Darkstar77, 2024-03-23:
@suchmememanyskill The last dev build worked when the issue happened. It reused the password, and reconnected much easier. I'll load up that latest build today, and keep you posted.

closed 2024-03-29 by suchmememanyskill