suchmememanyskill/CYD-Klipper #71: "Printer is not ready" Error

crazy-hair, 2024-03-22:
On the latest firmware (1.5) I'm getting an error screen like this every 15s-1m:

[image]

It stays on screen for about 5 seconds, but when it goes away it knocks you back out into the main screen, interrupting whatever I was currently doing.

suchmememanyskill, 2024-03-22:
Possibly caused by https://github.com/suchmememanyskill/CYD-Klipper/blob/master/CYD-Klipper/src/core/data_setup.cpp#L224?

Seems like a moonraker fault, but i'll remove that line and send you a build when i have the time to do so

suchmememanyskill, 2024-03-23:
Can you try the following build? https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8396463313

crazy-hair, 2024-03-24:
From about 5 minutes of testing, with the older firmware running on another screen, I got 3 popups on the old firmware and none on the new one. So far, so good.

closed 2024-03-29 by suchmememanyskill