suchmememanyskill/CYD-Klipper #72: Unclear "timing" indications

Misterke, 2024-03-22:
I don't really know a better way to explain my problem than through this image:
[image]
[image]
[image]

On the left you see the display of the printer itself, on which Klipper is displaying the print status.  Zooming in, you'll notice Klipper is there indicating 31% complete and 51' already elapsed.
On the right you see the CYD-Klipper progress screen for the same printer.  Strangely enough it is only reporting 14.57% complete and 43'  elapsed.  And a ridiculously high 4h15' remaining ...

I tried the different options in CYD-Klipper for the time estimate, but nothing came really close to the values reported by Klipper or KlipperScreen or Mainsail and I did not really notice much difference between "Percentage", "Interpolated" and "Slicer" for the Estimated Time setting.

Most surprising is that even the elapsed time seems way off, even though that should not really be an estimate as it's a measurable unit.  I did notice that when hovering the Total time in Mainsail, you get a popup with the same value as what CYD-Klipper reports mentioned as "Print" time (and also a "Difference" is mentioned in that popup which seems to reflect the delta between the Total time normally reported by Mainsail (and other GUIs) and this "Print" time.  Considering all other UIs I'm running report the thing Mainsail calls "Total Time", it would probably also make more sense for CYD-Klipper to report that same value rather than the "Print" time.

suchmememanyskill, 2024-03-22:
Percentage i'm unsure about; i use the value reported by moonraker, which should be the % of the file read.

Time is hard to estimate, especially when you have minimal info on the printer itself. It is not a known constant. 

The 'slicer' time estimate should only use your slicer's estimate. Please double check if it's read properly by checking the console log, it should report if it found a slicer estimate. Otherwise it'll silently fall back on a percentage based estimate.

(Another way to check is by reading (printer ip)/server/files/metadata?filename=(full file path from gcode root))

Misterke, 2024-03-22:
When actually attached to the printer and printing I don't really have a serial connection to it.  I tried getting the metadate from the server, but can't seem to find the correct full file path (my names have spaces and other special characters in them).  I do see the filament usage in the g-code files section of Mainsail, so normally I should have all the metadata OrcaSlicer adds.

But ... that isn't explaining why the elapsed time is wrong, right?  I had a look at the code and it seems you are using `printer.elapsed_time_s = status["print_stats"]["print_duration"];`, instead of `total_duration`.  I can understand that this value is used for remaining time calculations if the slicer is indicating the total time based on just the printing, but for presenting how much time has already passed to the user, I would expected `total_duration` is much more logical, no?  That would explain the difference in elapsed time between all other GUIs en CYD-Klipper.

While at it, I had a look at the code you use to determine the progress percentage.  As far as I can tell you are always just using `printer.print_progress = status["virtual_sdcard"]["progress"];`, correct?   If enabled in Klipper `["display_status"]["progress"] could be used as it provides the progress status the slicer adds via M73.  Another option would be to use `printer.current_layer / printer.total_layers` or better `printer.filament_used_mm / metadata["filament_total"]` if the metadata is available.  Maybe giving some of these alternatives as options similar to what you now have for the remaining time?  In fact, those could be coupled, since `elapsed/(elapsed+remaining) == progress`, so any option on how to estimate remaining time also is one on how to estimate progress percentage and vice versa.

suchmememanyskill, 2024-03-22:
> I do see the filament usage in the g-code files section of Mainsail, so normally I should have all the metadata OrcaSlicer adds.

This is meaningless if CYD-Klipper can't fetch the data. If it was able to, you can see a stark difference in estimated time if you switch between percentage, interpolated, and slicer. Can you double check this?

> But ... that isn't explaining why the elapsed time is wrong, right?

You are right, but for the wrong reason i believe. I am using [moonraker's docs](https://moonraker.readthedocs.io/en/latest/web_api/#basic-print-status) as a guideline how to calculate this. Using `total_duration` would balloon the wrong estimated time further out of proportions (as suddenly you're wasting 1-2 mins without going any % forward).

The real issue i believe is that i'm not sampling from the last 30 seconds or so, but the entire print. That i will likely never change due to memory concerns.

> ["display_status"]["progress"]

I can maybe add this down the line, sure, although it wouldn't help that much.

> printer.current_layer / printer.total_layers

This will cause the estimated time to not update until a layer completes. This doesn't seem like a great option.

> printer.filament_used_mm / metadata["filament_total"]

This will cause issues if doing prints that use progressively less filament the higher up it goes. There is no perfect option here. 

I do not know how Mainsail calculates the ETA. Can you try to look into Mainsail's source code for how exactly this is calculated?

Misterke, 2024-03-25:
@suchmememanyskill I tried out your latest dev version where you did add the display_status progress and added my own change (for which I've entered a pull-request) to get correct elapsed time and that gives me a display that is quite well in sync with all other GUIs I have ...

closed 2024-03-29 by suchmememanyskill