suchmememanyskill/CYD-Klipper #74: Custom buttons

szafranski, 2024-03-24:
Would it be possible to make buttons editable? I mean increments in the "move" tab etc? Btw - congrats on the project development. Awesome!

suchmememanyskill, 2024-03-24:
What reason do you have for making the buttons editable?

suchmememanyskill, 2024-03-24:
Would something like this be more suitable?

[image]

szafranski, 2024-03-25:
Thank you for quick response and quick action! So this is important since usually I move Z axis more than 10mm and I have to click the button few times. From the other hand, I would like to have some another values stored on button for X and Y. I think it would be best if you could adjust the stored button value by pressing it longer. Doable? :)

suchmememanyskill, 2024-03-26:
@szafranski can you try the following build?

How to run a developer build: https://github.com/suchmememanyskill/CYD-Klipper/blob/dev/Build.md
Firmware: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8443408965

It's not particulary efficient and i'll probably eventually refactor it, but for now it works.
In the move panel a pencil icon is now present. Pressing it toggles it's state and pressing any increment button will open the keyboard. Setting a value sets both the positive and negative counterparts. Values up to 999 are supported with one decimal point

szafranski, 2024-03-27:
Awesome! ~Tested. Totally works for me :) I will add another issues but not to be picky, just to help with the feedback. Thanks!

closed 2024-03-29 by suchmememanyskill