suchmememanyskill/CYD-Klipper #79: Screen usage without the Wi-Fi (direct connection)

szafranski, 2024-03-29:
Hi! Do you think it would be possible to make the screen work with direct connection?
Via spi, i2c or so?

suchmememanyskill, 2024-03-29:
Out of scope for the project. It would require both maintaining a 'server' that runs on the pi, that just forwards the moonraker calls to the cyd. On the display's end, the entire communication engine needs to be rewritten. I'm simply not interested in doing this.

If you want a more 'local' option, might be worth making your pi a hidden AP, then connecting to it using CYD-Klipper?

closed 2024-03-29 by suchmememanyskill

szafranski, 2024-03-29:
Sure. I will explore hotspot way, but I believe that pi cannot work as a client and a hotspot at the same time.