suchmememanyskill/CYD-Klipper #80: Files list available when printing

szafranski, 2024-03-29:
Would it be possible to make files list accessible during the print? Not to start the new print (of course),
just to view what files are present in the memory.

suchmememanyskill, 2024-03-29:
Why would you want to see what files you can print without actually being able to print them?

While yes, you can show the file list while printing quite easily, i really do not see the point. Especially when screen space is at a premium and showing one more button will probably make the ui more cluttered than it needs to be

szafranski, 2024-03-29:
To check if some file was already sliced and sent. I mean there is plenty of space near the pause button.

suchmememanyskill, 2024-06-28:
Implemented in latest release

closed 2024-06-28 by suchmememanyskill