suchmememanyskill/CYD-Klipper #81: Screen badly calibrated after initial setup

szafranski, 2024-03-29:
Hi. So I flashed the screen few times (for testing purposes etc). It required me to keep the button being pressed all the time during the flashing procedure. That part is not that bad. But because of that, right after the flashing, screen prompted me to calibrate it - by clicking on the points. Few times I accidently pushed the screen in the wrong place - since I was holding the boot button previously - and screen was unoperable after this - cliks were not registered for obvious reason. I could not connect to Wi-Fi then and I had no way to calibrate it again. I had to reflash it. Would it be possible to make some dumb-proof way of preventing it? Like maybe some timeout when somone won'tchoose Wi-Fi network in like 10 secs or so to go back BEFORE the calibrating screen procedure? Or maybe if you reset the screen before selecting the Wi-Fi network it should go BEFORE the calibrating procedure as well? Thanks!

suchmememanyskill, 2024-03-29:
I have played around with this idea before, but it's really annoying to find a 'trigger' for it. I don't think waiting 10 seconds is a good trigger.

Possibly holding the 'BOOT' button for x amount of seconds while the display does nothing is a good substitute, but that'll require some refactoring

szafranski, 2024-03-29:
Sounds like a good solution possibly. BTW - I wrote those issues to help. I appreciate your awesome work!

suchmememanyskill, 2024-03-29:
Can you try this build?
https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8484695500

When holding the BOOT button for 8 seconds, the screen will restart into touch calibration. Only on the 2.8" resistive model

closed 2024-03-29 by suchmememanyskill