suchmememanyskill/CYD-Klipper #84: Layer popup overlaping.

RH3D, 2024-04-05:
The layer information popup on top of the screen is overlaping the cross/close button (top right corner) for settings like fan, speed, etc. This means I have to wait for it to disappear or I need to click on the very left edge where I click on the main buttons.
In the worst case with short layer times, the popup basicaly doesn't go away so I have no chance to click the close button.

To me, this popup is not really needed as there is enough information about the current layer already.
I am using the 2,8" version ESP32-2432S028R with latest OTA FW.
[image]

suchmememanyskill, 2024-04-05:
This seems like your own fault; It seems like M117 is called with a message (https://www.klipper3d.org/G-Codes.html). This message gets displayed on the CYD. If you spam messages, it makes sense that those messages appear in rapid succession.

Might be worth having an option to disable messaging entirely, though

suchmememanyskill, 2024-04-05:
Would you mind trying the following build?

Build: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/8575685113
Instructions for running build: https://github.com/suchmememanyskill/CYD-Klipper/blob/master/Build.md

RH3D, 2024-04-05:
Oh! Thank you, you are totally right about the M117, it was probably a relic from my old Marlin profiles but I really don't need it now with Klipper so disabling it is enough.

I will try the build and post an update when I get to test it.
I like both the dismiss the popup by tapping and toggle in settings.
Thank you!

closed 2024-04-14 by suchmememanyskill

RH3D, 2024-04-16:
Sorry for my late reply, but I've been fairly busy last days and wanted to test the update on more then just a few prints.

So far everythings has been working as expected and I didn't notice any issues with it.
Thanks for the implementation again!