suchmememanyskill/CYD-Klipper #85: Web upload not work on my slightly odd CYD

alanvgreen, 2024-04-12:
Hi: I bought a CYD from Aliexpress [link to AliExpress](https://www.aliexpress.com/item/1005006160645147.html) for the purpose of installingCYD-Klipper, and after looking at for a bit, I can see that, despite having `ESP32-2432S028` it is _not quite_ the same as some of the other pictures I've seen online. For example, it has a USB-C port in addition to the micro USB.

[image]

I was not able to make the web installer work - it fails with a "Failed to initialize" error.  Here is the content of the JS console log
[logs.txt](https://github.com/suchmememanyskill/CYD-Klipper/files/14957904/logs.txt)

I was able to upload v1.6.0 using `esptool.exe`, and it works just fine, excepting a FPS and CPU overlay in the bottom right corner that sometimes gets in the way of important text.
[image]

suchmememanyskill, 2024-04-12:
I have the same unit, and can upload to it just fine via the web ui. I suspect it's not the screen

The fps counter only shows if you use a developer build. Did i upload a developer build by accident as release v1.6.0?

suchmememanyskill, 2024-04-12:
Just checked, release v1.6.0 contains a release build of v1.6.0

alanvgreen, 2024-04-13:
> I have the same unit, and can upload to it just fine via the web ui. I suspect it's not the screen

After rebooting windows, it works!

> The fps counter only shows if you use a developer build

Yes, I did you a developer build. I was following the instructions in `build.md`... which say, in large letters, "Developer build".

It's working better than I had hoped. Thank you for the amazing product and thank you too for following up on this issue.

closed 2024-04-13 by alanvgreen

suchmememanyskill, 2024-04-13:
Small tip btw, before i forget:

That type of screen should use the 'inverted colors' and 'color fix' settings. By default the colors are wonky and inverted