suchmememanyskill/CYD-Klipper #88: Bootloop while/on connecting to moonraker

szafran81, 2024-04-14:
Hi,

I've bought two of ESP32-2432S028 to check how this project would work (and if it would I plan to move to bigger screen).
I've flashed the project on both of the screens some time ago (don't remember - month or two). And I've got a bootloop, right after the screen showed "Connecting to Klipper".

https://github.com/suchmememanyskill/CYD-Klipper/assets/2608332/f28c2345-2aba-4ccc-9a65-c8ecf01d6476

Now I've reflashed a fresh v1.6.1 with data format.

This is what I'm getting now:

https://github.com/suchmememanyskill/CYD-Klipper/assets/2608332/33dfddde-a730-4cd8-9fc4-f76233789372

This may help:

[image]

Software versions installed on the Klipper host PC:
[image]

What can I do to make the screens to work properly?

suchmememanyskill, 2024-04-14:
Could you build via platformio? https://github.com/suchmememanyskill/CYD-Klipper/blob/master/Build.md#building-via-platformio

One thing that this doc doesn't mention is that you also have to install [git](https://git-scm.com/download/win)

szafran81, 2024-04-15:
> (specifically build esp32-2432S028R)
That tells me nothing.

I've installed git and vscode (btw typical Microsoft... you download an installer, and run it and it says that it should not be run as Admin - it wasn't... and it still installs and after running says it's running with admin rights etc. M$ crap -- sorry for the OT)

After VSCode installing everything, and me opening the project I get this:
[image]

Then I read this:
[image]
Say what?

Then this:
[image]
Done.

Then this:
[image]
Where? What? How?

Then I pressed "Build" button... Oh crap that was a lot of terminal output, installing things compiling etc.

Then I've got this:
[image]

Don't really know what to do next?

suchmememanyskill, 2024-04-15:
Good point, might want to add some screenshots to the build guide.

You didn't click on the bug icon on the left side; this changes the side panel to display a list of devices instead of files. You can expand one of these to reveal the upload & monitor button

szafran81, 2024-04-15:
Ah... my mistake... I did in fact click on it, but somehow it got back to the folder view - which I didn't realize.
Just checked remotelly on my PC, and now I can see the list of devies:

[image]

Now I'm at work, so I can't test it. Will do so today after getting back, or tomorrow morning, and will get back to you.

P.S.
Screenshots in the guide would help a lot to someone who doesn't use VSCode (well I do use it, but only as config editor on my Home Assistant install).

szafran81, 2024-04-15:
esp32-2432S028R:

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Hello World
Config version: 6
Screen init done
Connecting to Skajniet_IoT with a password length of 22
OTA Update Result: -3
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008ab21  PS      : 0x00060130  A0      : 0x800d3f55  A1      : 0x3ffb2030  
A2      : 0x3ffc455c  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb2000
A10     : 0x3ffc455c  A11     : 0x3f400385  A12     : 0x3ffb20e0  A13     : 0x000005dc
A14     : 0x7ff00000  A15     : 0x7ff77000  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008ab21  LEND    : 0x4008ab35  LCOUNT  : 0xffffffff

Backtrace: 0x4008ab1e:0x3ffb2030 0x400d3f52:0x3ffb2040 0x400d4405:0x3ffb2230 0x400d6848:0x3ffb2260 0x4010b336:0x3ffb2290

  #0  0x4008ab1e:0x3ffb2030 in strcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strcpy.S:98
  #1  0x400d3f52:0x3ffb2040 in fetch_printer_data() at src/core/data_setup.cpp:215
  #2  0x400d4405:0x3ffb2230 in data_setup() at src/core/data_setup.cpp:445
  #3  0x400d6848:0x3ffb2260 in setup() at src/main.cpp:24
  #4  0x4010b336:0x3ffb2290 in loopTask(void*) at C:/Users/Szafran/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 9fe92b87dfb02fcf

Rebooting...
```

esp32-2432S028Rv1-SD:
```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Hello World
Config version: 6
Screen init done
Connecting to Skajniet_IoT with a password length of 22
OTA Update Result: -2
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008aeed  PS      : 0x00060130  A0      : 0x800d4125  A1      : 0x3ffb2030  
A2      : 0x3ffc469c  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb2000  
A10     : 0x3ffc469c  A11     : 0x3f400385  A12     : 0x3ffb20e0  A13     : 0x000005dc
A14     : 0x7ff00000  A15     : 0x7ff77000  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008aeed  LEND    : 0x4008af01  LCOUNT  : 0xffffffff

Backtrace: 0x4008aeea:0x3ffb2030 0x400d4122:0x3ffb2040 0x400d45d5:0x3ffb2230 0x400d6914:0x3ffb2260 0x40109996:0x3ffb2290

[... 68 lines cut ...]
OTA Update Result: -2
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008aeed  PS      : 0x00060130  A0      : 0x800d4139  A1      : 0x3ffb2030  
A2      : 0x3ffc465c  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb2000  
A10     : 0x3ffc465c  A11     : 0x3f400385  A12     : 0x3ffb20e0  A13     : 0x000005dc
A14     : 0x7ff00000  A15     : 0x7ff77000  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008aeed  LEND    : 0x4008af01  LCOUNT  : 0xffffffff

Backtrace: 0x4008aeea:0x3ffb2030 0x400d4136:0x3ffb2040 0x400d45e9:0x3ffb2230 0x400d6928:0x3ffb2260 0x4010945e:0x3ffb2290

  #0  0x4008aeea:0x3ffb2030 in strcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strcpy.S:98
  #1  0x400d4136:0x3ffb2040 in fetch_printer_data() at src/core/data_setup.cpp:215
  #2  0x400d45e9:0x3ffb2230 in data_setup() at src/core/data_setup.cpp:445
  #3  0x400d6928:0x3ffb2260 in setup() at src/main.cpp:24
  #4  0x4010945e:0x3ffb2290 in loopTask(void*) at C:/Users/Szafran/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: dad3b8bc7603dfee

Rebooting...
```

The last gave me a flipped mirrored image with orange button. Others behaved just like the one uploaded via web.

suchmememanyskill, 2024-04-15:
Please only run `esp32-2432S028R`, not the smartdisplay variants (they are meant for specific variations of the screen and are developer builds only).

I have pushed a proposed fix to the [dev branch](https://github.com/suchmememanyskill/CYD-Klipper/tree/dev). Please give it a go

szafran81, 2024-04-15:
> Please only run `esp32-2432S028R`

No problem. Just had some time and wanted to play with that ;)

Now with the new commit:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Hello World
Config version: 6
Screen init done
Connecting to Skajniet_IoT with a password length of 22
OTA Update Result: -3
Heap space pre-file-parse: 136400 bytes
Json parse: Ok
Heap space post-file-parse: 134172 bytes
Got 0 files. Request took 24ms, parsing took 1ms
```
and...

[image]

so I think it worked.

suchmememanyskill, 2024-04-15:
> Will I be able to do on device update to a future version when it comes out or will I have to reflash it via PC again?

Yeah, whenever i make a new release you should be able to update OTA (well, assuming you see v1.6.1 or below in settings)

szafran81, 2024-04-15:
It looks like this right now:
[image]

suchmememanyskill, 2024-04-15:
Ah. I guess you'd need to manually press that button once it says v1.6.2 or up

szafran81, 2024-04-16:
Great. Thank you.

closed 2024-04-16 by szafran81