suchmememanyskill/CYD-Klipper #89: Ability to make folders in macros menu

szafranski, 2024-04-14:
Hi! It's me again ;) I would like to request the ability to make folders in macros menu. I can explain more if needed. Thank you for the awesome work!

suchmememanyskill, 2024-04-15:
Why do you need so many macros, out of curiosity?

Also this would be very annoying to implement, and would require some non standard data (like some foldername in a macro name). Not a huge fan of implementing this

szafranski, 2024-04-15:
Actually you maybe right. I like to have all my macros at hand, when near
the printer but it is nor required actually. What about ordering them with
alphabetical order at least then?

Pawel

suchmememanyskill, 2024-04-15:
Do they not show up in the same order as they are placed in the printer.cfg file?

szafranski, 2024-04-22:
Oh, yes, they are in the same order as in the config. Could there be the option to sort them alphabetically in the menu or sth then? :)

suchmememanyskill, 2024-06-28:
Implemented in latest release

closed 2024-06-28 by suchmememanyskill