suchmememanyskill/CYD-Klipper #92: Not all data is displayed on the screen

szafran81, 2024-04-20:
Hi,

After recent screen to moonraker connection problems fixed I've had some idle time on my two printers. Now they are printing and I can finally do some test with the screens.
Ehat I've seen so far... The screen always shows that the printers are idle. I'm getting empy file list all the time. The Z height and temperatures seem to be displayed ok.
Since the display thinks the printers are idling it switches off even if the option to not turn off while printing is turned on.

Any help with fixing the above would be appreciated.

suchmememanyskill, 2024-04-20:
> I'm getting empy file list all the time.

Note if you have too many files stored on the printer CYD-Klipper may fail to load them due to memory limitations

-----

Can you get me the logs of CYD-Klipper while it should be reading print progress info?

Can you also, while you're printing, get the output of `http://{PRINTER_IP}:{PRINTER_PORT}/printer/objects/query?extruder&heater_bed&toolhead&gcode_move&virtual_sdcard&print_stats&webhooks&fan&display_status` in a web browser (right click > save as)

Maybe a little more sensitive as it includes file names, but same with `http://{PRINTER_IP}:{PRINTER_PORT}/server/files/list`

suchmememanyskill, 2024-04-20:
[image]
Might be important also

szafran81, 2024-04-20:
[image]

I'm failry up to date - so I don't think that's the problem.

> Note if you have too many files stored on the printer CYD-Klipper may fail to load them due to memory limitations

Printer in question has 1 folder and 3 gcode files in my OP file list (while typing this I've realized that my klipper gcode folder is probably different to my OctoPrint gcode folder, and that might be why the file list is empty).

> http://{PRINTER_IP}:{PRINTER_PORT}/server/files/list

```
{"result": []}
```

> http://{PRINTER_IP}:{PRINTER_PORT}/printer/objects/query?extruder&heater_bed&toolhead&gcode_move&virtual_sdcard&print_stats&webhooks&fan&display_status

```
{"result": {"eventtime": 55477.314883225, "status": {"extruder": {"temperature": 255.46, "target": 255.0, "power": 0.3326363968369232, "can_extrude": true, "pressure_advance": 0.03, "smooth_time": 0.04, "motion_queue": null}, "heater_bed": {"temperature": 109.72, "target": 110.0, "power": 0.21992977122220528}, "toolhead": {"homed_axes": "xyz", "axis_minimum": [-3.0, -1.0, -3.0, 0.0], "axis_maximum": [298.0, 307.0, 249.0, 0.0], "print_time": 55499.812156014654, "stalls": 3, "estimated_print_time": 55496.81965964445, "extruder": "extruder", "position": [245.574, 160.663, 24.0, 111144.18056002347], "max_velocity": 400.0, "max_accel": 5000.0, "minimum_cruise_ratio": 0.5, "square_corner_velocity": 5.0}, "gcode_move": {"speed_factor": 1.0, "speed": 6000.0, "extrude_factor": 1.0, "absolute_coordinates": true, "absolute_extrude": false, "homing_origin": [0.0, 0.0, -0.38, 0.0], "position": [245.574, 160.663, 24.02, 111144.18056002347], "gcode_position": [245.574, 160.663, 24.4, 454.3645399954985]}, "virtual_sdcard": {}, "print_stats": {}, "webhooks": {"state": "ready", "state_message": "Printer is ready"}, "fan": {"speed": 0.5, "rpm": null}, "display_status": {"progress": 0.91, "message": "ETA is 15:32:05"}}}}
```

> Can you get me the logs of CYD-Klipper while it should be reading print progress info?

How can I do that?

suchmememanyskill, 2024-04-20:
the "virtual_sdcard" and "print_stats" printer objects are empty. These need to be filled in order for CYD-Klipper to detect status updates. This is also why you got that nullref exception earlier. I'm unsure how these printer objects work when using Octoprint besides Klipper. 

Might be worth setting up a frontend like Mainsail. If Mainsail works, CYD-Klipper should also work. Might also be as simple as adding the following to printer.cfg:

```
[virtual_sdcard]
path: ~/printer_data/gcodes
on_error_gcode: CANCEL_PRINT

[display_status]
```

szafran81, 2024-04-20:
> (Why would you even want to use Octoprint when you have Mailsail or Fluid??)

Because IMO it's better in almost every way than both of them combined. Most of all none of them support plugins. I use a few of of them, but they help a lot (like spool management etc.). And yes - they both work. I've just removed them after testing.

> [display_status]

I have ```[display]``` on both printers, so ```[display_status]``` is available by default (according to Klipper manual).

> [virtual_sdcard]
> path: ~/printer_data/gcodes

I've added this, but have to wait for the print to finish, so I can restart Klipper.

> This is also why you got that nullref exception earlier.

Maybe. But this works with moonraker, so shouldn't it work independently of installed/used front end?

Will write some more after the print is done, and I can test the changes...

suchmememanyskill, 2024-04-20:
So does the screen fully work now or?

szafran81, 2024-04-20:
> Also spoolman is built into fluidd and mainsail ;)

Yeah, maybe. But that was just an example. I use more...
[image]
...and the list goes on...

> So does the screen fully work now or?

File list works.
Current status works if I run a print from the screen, or the printers LCD.

It still shows Idle when a print is started from OP (so the only way that I start a print). It would be nice if it showed the proper info while printing from OP.

suchmememanyskill, 2024-04-20:
As long as Moonraker doesn't know about the print status, i can't really do a whole lot about this

szafran81, 2024-04-20:
How about using:

```
"display_status": {"progress": 0.0, "message": "Rozgrzewka..."}
```
and reacting to ```progress > 0```?

And from what I've seen there are more differences in status display between idle and printing states.
Now I've started another print. I'll compare it tomorrow when it finishes.

suchmememanyskill, 2024-04-20:
> and reacting to progress > 0?

I wouldn't be able to tell if a print is paused or not

szafran81, 2024-04-20:
> I wouldn't be able to tell if a print is paused or not

That is an issue. No doubt. At this time I don't have any ideas on how to fix that.

But I think that at least printer controls should be locked when progress > 0. My previous print was ended with a "move out of bounds" error because I accidentially clicked ```+10``` on the controls screen. Good thing it was less than 10min to the end of the print and it was a functional print, so it's usable - in other case a ~16h print would be wasted.

Will get back to you tomorrow after current print ends...

suchmememanyskill, 2024-04-20:
> That is an issue. No doubt. At this time I don't have any ideas on how to fix that.
>
> But I think that at least printer controls should be locked when progress > 0. My previous print was ended with a [...]

I feel like this is out of scope for the project. The screen is a frontend for Moonraker/Klipper, not Octoprint puppeting Klipper. It might be worth using a Octoprint-native screen instead

szafran81, 2024-04-20:
> I feel like this is out of scope for the project. The screen is a frontend for Moonraker/Klipper, not Octoprint puppeting Klipper.

As you wish.
As for me - the project is of no use in my case if no changes are made. It's just dangerous to use in connection with OP/Klipper duo.

But, please, add info on the main project page that's it's incompatible with such a configuration for anyone else using OP and wanting a remote control for their printer.

> It might be worth using a Octoprint-native screen instead

That's what I'm going to look into now.

Thank you for your time, and help.

closed 2024-04-20 by szafran81