suchmememanyskill/CYD-Klipper #98: ESP32-8048S070C support ?

Ramsy504, 2024-05-05:
Hello will the ESP32-8048S070C Version supported ?

suchmememanyskill, 2024-05-05:
I don't think it's a good idea to use that type of screen; the ui will be incredibly slow (~4fps)

dronecz, 2024-05-06:
It is not so bad as you can see [here](https://youtu.be/rIf76XW0b0I). 

There are still some issues:
 - flickering at the start, when board is connecting to WiFi
 - very quick restarts until set_screen_brightness() function is disabled

My config and changes are in [dev](https://github.com/suchmememanyskill/CYD-Klipper/commit/13a9f318d25c3c7319721e478778ee73d0d04b71) branch of my fork.

Ramsy504, 2024-06-01:
Looks great, Do you have a ready to Install Version?

suchmememanyskill, 2024-06-28:
@dronecz when i was browsing the xtouch repo i noticed they have some sort of a boot delay. If you add a sleep method in the main function for 10-30s or so, does it boot normally afterwards without any hacks like this?

dronecz, 2024-06-28:
@suchmememanyskill Can you please point me to the code you have in mind? Do you mean repo for GT911 touch driver or something else?

suchmememanyskill, 2024-06-28:
@dronecz 

Seems to be put after screen init and before wifi connect/menu loading

https://github.com/xperiments-in/xtouch/blob/main/src%2Fmain.cpp#L50

You can sleep the device using the delay(int ms) function. It sleeps for the given time in miliseconds

dronecz, 2024-06-28:
@suchmememanyskill Xtouch driver is for resistive panels, but my display has a capacitive panel, so I do not think the code you are referring to is valid for my display.

suchmememanyskill, 2024-06-28:
@dronecz this isn't display driver code. I don't know where you got that from. I suspect slowly initializing components will help

dronecz, 2024-07-02:
@suchmememanyskill Oh, sorry, I did not understood what you mean, but I now know what you mean. I will look at the code when I will have time. Thx

Saxin, 2024-11-26:
@suchmememanyskill  thank you a lot for the hard work and for this project given to the community.
I've also have this board ESP32-8048S070C, wondering why you say it's not suitable for KYC ?

> I don't think it's a good idea to use that type of screen; the ui will be incredibly slow (~4fps) 

Thanks again, Eddi