suchmememanyskill/CYD-Klipper #101: Macros for Extrude/Retract.

FeromonDE, 2024-05-23:
[image]
Hello, it would be nice to be able to add your own macros for Extrude/Retract to these two buttons.

suchmememanyskill, 2024-05-24:
What use case do you have for then? Might be able to implement that natively rather than hack around it with a macro

FeromonDE, 2024-05-24:
My macros do everything in one click, but if you use these buttons, you first have to wait until the extruder warms up, this is a 2 step process.

suchmememanyskill, 2024-05-25:
@FeromonDE 

Please test the following build:
Build: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/9234905349
Install instructions: https://github.com/suchmememanyskill/CYD-Klipper/blob/dev/Manual_Install.md

FeromonDE, 2024-05-25:
i can not download the Firmware.

`esp32-3248S035C  FAILED    00:01:19.380
==================== 1 failed, 0 succeeded in 00:01:19.380 ====================
Traceback (most recent call last):
  File "/home/runner/work/CYD-Klipper/CYD-Klipper/ci.py", line 67, in <module>
    subprocess.run(["pio", "run", "-e", port], check=True)
  File "/opt/hostedtoolcache/Python/3.9.19/x64/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['pio', 'run', '-e', 'esp32-3248S035C']' returned non-zero exit status 1.
Error: Process completed with exit code 1.`

suchmememanyskill, 2024-05-25:
@FeromonDE I'm terribly sorry, i should have waited the duration of the action build before sending it to you. Inconviniently, expressif updated their framework toolchain which caused this issue. I have now locked the version to the same version i run locally

Build: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/9238631767

FeromonDE, 2024-05-25:
I installed this version, the use of the CPU and the number of FPS in the lower right corner are now shown on the screen. How to rename my macros for Retract / Extrude buttons?

suchmememanyskill, 2024-05-26:
@FeromonDE in the settings a new toggle was added called 'Custom Filament Move Macros'.

Instead of calling a predefined move instruction, it calls a predefined macro (FILAMENT_RETRACT and FILAMENT_EXTRUDE)

FeromonDE, 2024-05-26:
@suchmememanyskill , I tested this version, everything works as it should. Having previously renamed my macros. Thank you.

closed 2024-05-28 by suchmememanyskill