suchmememanyskill/CYD-Klipper #102: Programming ESP32-3248S035C with Arduino C++ code

eagl1, 2024-05-25:
Hello,

I just uploaded your install code through this page:
[https://suchmememanyskill.github.io/CYD-Klipper/](url)

So I guess this code is done in Python. And I tried to program it in Arduino IDE, but it's not working.

I set all the files using TFT_espi and lvgl, no example code worked.

Do you have any idea of what could be the problem ?

suchmememanyskill, 2024-05-25:
I'm very confused, what exactly is your issue?

'This code is done in python' the CI is done in python, CYD-Klipper itself in c++

I also don't use the Arduino IDE, i use platformio

eagl1, 2024-05-25:
My issue is that I want to get the correct source code to program this board with Arduino IDE, I tried VS code I faced some difficulties and platformio doesn't work and freeze.

My ST7796 driver works well with TFT_espi library and my 4" ST7796 with XPT2046 resistive driver.

I set the SPI pins accordingly but it won't work.

suchmememanyskill, 2024-05-25:
Are you asking for general programming help or help with getting my program running in Arduino IDE?

For either case, i offer no support

eagl1, 2024-05-25:
The second request. I want to know where is the problem that the code on Arduino IDE isn't working. And I just programmed my board with your code, so I want to learn your source code.

suchmememanyskill, 2024-05-25:
I only support platformio. Using the Arduino IDE is out of scope for this project.

closed 2024-05-25 by suchmememanyskill

eagl1, 2024-05-25:
Is there any chance to compile the project on VS code ? because platformio doesn't work with me and freezes up.

eagl1, 2024-05-25:
I tried to compile the code in VS code.

I've put your run code in .ino file:

[image]

One of my main problems is that I get this `[Warning] Failed to generate IntelliSense configuration.`

eagl1, 2024-05-25:
I'm sorry but I have a question, where is `CYD_SCREEN_GAP_PX` is defined ? I searched for it but I couldn't find it.

eagl1, 2024-05-25:
OK I found it in the json file.

Thanks :)