suchmememanyskill/CYD-Klipper #105: ESP32-8048S043C not programming and not stable usb connection

eagl1, 2024-05-31:
Hi,

I have both `ESP32-3248S035C` & `ESP32-8048S043C`.

I was able to program `ESP32-3248S035C` easily on your web installer, but `ESP32-8048S043C` isn't programming.

But the `ESP32-8048S043C` is actually not stable in the usb connection, after a minute, it disappears from the usb connection when I try to program it, but it's display in the COM port list in the windows device manager.

It doesn't program on any platform, arduino ide, your web installer, openHAS web installer. I don't know why.

suchmememanyskill, 2024-05-31:
This screen's CI build has some issues that i've yet to figure out (see #67)

For now, please try manually installing via platformio: https://github.com/suchmememanyskill/CYD-Klipper/blob/master/Manual_Install.md

eagl1, 2024-05-31:
I tried that:

[image]

The board is working and running the demo code came from the manufacturer but I can't program it.

suchmememanyskill, 2024-06-01:
This is however, not what i asked you to do. I asked you to install via platformio, not via esptool...

Scroll down in the doc that i sent

(Btw, i suspect you're missing a serial driver somewhere)

eagl1, 2024-06-01:
I installed platformio and opened the project folder.

[image]

I didn't get the menu like the one you posted in the tutorial.

suchmememanyskill, 2024-06-01:
Again, please see the instructions.

You didn't click the bug/alien icon on the left side. This is (1) in the screenshot at the bottom of the instructions

eagl1, 2024-06-01:
When I do this:

[image]

I get this:

[image]

And it asks me this in the output window:

```
Resolving esp32-2432S028R dependencies...
Platform Manager: Installing git+https://github.com/platformio/platform-espressif32#v6.4.0
UserSideException: Please install Git client from https://git-scm.com/downloads
```

Even I've already installed git

eagl1, 2024-06-01:
And even when I again click on the alien/bug I get this project tree:

[image]

eagl1, 2024-06-01:
OK, wait a minute ! I think I found the source of the problem.

The problem is that I installed VS code in another location other than the default one.

So I uninstalled and re-installed it as is, and I think it's working now and it automatically started the unpack your project :)

[image]

suchmememanyskill, 2024-06-01:
So can you open the folder for your screen and select upload & monitor?

eagl1, 2024-06-01:
Yep, but the same port problem:

[image]

The board is booting every time with the manufacturer lvgl demo, I couldn't program the board with anything else until now.

suchmememanyskill, 2024-06-01:
Actually, weird question. Are you holding down the BOOT button while the screen is flashing on and off (so while it says connecting....) in the terminal?

This is stated both on my install page and in my docs, just want to make sure

eagl1, 2024-06-01:
I did both methods with and without pressing on the boot button.

```
Building in release mode
Retrieving maximum program size .pio\build\esp32-8048S043C-SD\firmware.elf
Checking size .pio\build\esp32-8048S043C-SD\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  28.2% (used 92540 bytes from 327680 bytes)
Flash: [==        ]  18.0% (used 1177825 bytes from 6553600 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa   
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM6
Forcing reset using 1200bps open/close on port COM6
Waiting for the new upload port...
Uploading .pio\build\esp32-8048S043C-SD\firmware.bin
esptool.py v4.5.1
Serial port COM6
Connecting...

A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Access is denied.', None, 5)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 1
=================================== [FAILED] Took 20.87 seconds ===================================
Environment         Status    Duration
------------------  --------  ------------
esp32-8048S043C-SD  FAILED    00:00:20.869
============================== 1 failed, 0 succeeded in 00:00:20.869 ==============================
 *  The terminal process "C:\Users\eagle\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--target', 'monitor', '--environment', 'esp32-8048S043C-SD'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

It detects the board but can't program it.

suchmememanyskill, 2024-06-01:
PermissionError(13, 'Access is denied.', None, 5) might be worth looking into. This error message usually means, at least from my side, that the device is in use. Meaning you have the screen connected elsewhere, possibly?

Either way, this does not seem to be a specific issue with my project, but rather a misconfiguration on your end

eagl1, 2024-06-01:
Oh yeah absolutely, it's not an issue in your project. I tried to program it on different platforms, your method, arduino ide and the flash tool. No one worked.

I don't know what could be the problem, could it that there's no communication between the esp32 and the serial chip.

eagl1, 2024-06-01:
And also thank you for your help even with platformio, that I couldn't get it to work until today, now I can program my board through vs code and platformio. That would be much faster than arduino ide.

eagl1, 2024-06-06:
Who is the manufacturer of this display ?

The 4.3" I have isn't accessible in anyway, I tried to catch it with linux ubuntu in virtual machine but the same result.

I want to communicate with the manufacturing company and check what things I can do to resolve the usb connection issue.

eagl1, 2024-06-08:
OK, it worked :)

I installed an old CH340 driver following this post:
[https://github.com/espressif/esptool/issues/682](url)

closed 2024-06-08 by eagl1