suchmememanyskill/CYD-Klipper #116: Support for Elecrow 3248R035

Simone-Ludovici, 2024-07-05:
Hello suchmememanyskill,

I had a 3.5" ESP32-WROOM display with resistive touch, similar to the Sunton display but produced by Wizee and sold by Elecrow (now Elecrow has changed the name of the product but it should be the same display, https://www.elecrow.com/esp32-display-3-5-inch-hmi-display-spi-tft-lcd-touch-screen.html).

I am attaching the relevant files for the board and for the device if you are interested.
Thank you for your work.

Board
[esp32-3248R035.json](https://github.com/user-attachments/files/16113820/esp32-3248R035.json)
Device
[ESP32-3248R035.cpp.txt](https://github.com/user-attachments/files/16113824/ESP32-3248R035.cpp.txt)

MadRajib, 2024-11-30:
Hi @Simone-Ludovici  can you try https://github.com/MadRajib/CYD-Klipper-CrowPanel but its for 2.8R you can modify the json file it should work

MadRajib, 2024-12-02:
Hi @Simone-Ludovici  added 3,5 inch files https://github.com/MadRajib/CYD-Klipper-CrowPanel/tree/ESP32_CROWPANEL_35R, can you check once if it helps ?

pberki, 2025-03-08:
I tried to build  for esp32-CROWPANEL-35R (Projekt tasks - esp32-CROWPANEL-35R - Build) from https://github.com/MadRajib/CYD-Klipper-CrowPanel/tree/ESP32_CROWPANEL_35R, but the build fails with errors in data_setup.cpp

MadRajib, 2025-08-19:
Hi @pberki , could you share the error logs as well ?