suchmememanyskill/CYD-Klipper #118: Failed to connect

donutboy3, 2024-07-08:
It was working fine for a few days. Today it wouldn't connect so reflashed the CYD. I was able to enter my home wifi network information fine and when I entered my printers address it would respond with failed to connect. Any ideas? Thanks

suchmememanyskill, 2024-07-08:
Can you connect using a browser? Console logs of cyd klipper?

(Specifically http://{printer ip}/printer/info in a browser)

donutboy3, 2024-07-08:
I can connect to klipper through chrome browser and orca slicer. Here is the log information you requested. Thanks.

{
  "result": {
    "state": "ready",
    "state_message": "Printer is ready",
    "hostname": "hevo",
    "klipper_path": "/home/donutboy3/klipper",
    "python_path": "/home/donutboy3/klippy-env/bin/python",
    "process_id": 518,
    "user_id": 1000,
    "group_id": 1000,
    "log_file": "/home/donutboy3/printer_data/logs/klippy.log",
    "config_file": "/home/donutboy3/printer_data/config/printer.cfg",
    "software_version": "v0.12.0-260-g00cb683d",
    "cpu_info": "4 core ?"
  }
}

suchmememanyskill, 2024-07-09:
Your klipper printer and CYD-Klipper (and your pc) are all on the same network?
Do you have any blocking/authorization in place on the Klipper side (filtering ips, for example)?

closed 2024-07-14 by suchmememanyskill