suchmememanyskill/CYD-Klipper #120: First printer "offline" when adding second printer

jangrewe, 2024-07-14:
Dude, this is so awesome! I came across it while searching for projects for the CYD, and luckily had a spare CYD to use for you project - because that's just perfect use of this little screen.

Anyways, my issue: I've set it up for my Voron 2.4, and everything was fine, then i discovered that it can handle multiple printers, so i also added my Voron 0.2. Unfortunately the Voron 2.4 is now always shown as "Offline" (even though it's online 24/7) and i'm "In Control" over the Voron 0.2, with the "Control" button for the Voron 2.4 being disabled.

[image]

I'm not sure how to fix this, other than to reset the device and only add one printer... and get a second CYD. ;-)

Oh, this is a CYD2USB (as it's often referred to, because of the 2 USB connectors) with a ILI9342 driver, and i had to invert the colors - maybe that can be set as the default for these boards?

And a final Feature Request: Either a full serial shell or, even better, a simple web interface to manage the printers. Entering API keys via the on-screen keyboard isn't much fun... 😁

suchmememanyskill, 2024-07-14:
Hi,

First of all, thanks a lot for the coffees!

Second, you seem to actually have ran into a bug. I don't pass authententication through on the minimal requests done for multi printer mode. I have prepared a build that should fix this issue. Would you mind giving it a go?

Instructions: https://github.com/suchmememanyskill/CYD-Klipper/blob/master/Manual_Install.md
Build: https://github.com/suchmememanyskill/CYD-Klipper/actions/runs/9929717077

Third;

> maybe that can be set as the default for these boards?

I'm unsure if i can detect the difference in boards, as the same display driver is used across all 2.8" versions. That's why the toggle is there, though.

> a full serial shell 

Someone was ahead of you on that idea, see #117 :)

jangrewe, 2024-07-14:
Howdy, those are well deserved!

> I have prepared a build that should fix this issue. Would you mind giving it a go?

I just did, and it's working! Thanks for that quick fix 🥳

No worries about the boards, i just noticed that you did have a couple of variants the same board, so i thought this was already somehow sorted out, but with the wrong default... nevermind then, it's easy enough to configure (another example of the awesome job you're doing with this).

> see #117 

Great minds think alike! 😁
I'll have my fingers crossed that the PR gets merged before you release a new version, because i'd love to avoid entering the API key via the screen...
Oh, but i could enabled OTA, that would retain my config, right?

And maybe this is of any help to you, e.g. for the instructions how to flash a dev image: i'm typically using https://esp.huhn.me/ to quickly test any images without having to build them, or fire up esptool. It works great if you know to flash to `0x0`. ;-)

Thanks again!

suchmememanyskill, 2024-07-14:
> Oh, but i could enabled OTA, that would retain my config, right?

Updating OTA usually retains your configuration, except if i make changes to the configuration layout (example: i add an option in a breaking way). If there is any breaking changes, i specify this in the release notes.

Also, yeah, releasing this together with the serial configuration is the plan

jangrewe, 2024-07-14:
Awesome, looking forward to it! Already ordered a 2.5" with capacitive touch, and then i'll design a nice case with a 18650 and USB charging.

suchmememanyskill, 2024-08-02:
Released

closed 2024-08-02 by suchmememanyskill