suchmememanyskill/CYD-Klipper #121: Failed to connect

fylhtq7779, 2024-07-25:
Hi, I'm trying to connect with CYD to my printer, but I get the error "Failed to connect"
Sometimes the connection happens, I get into the CYD interface, but after a couple of seconds the connection disappears
I tried adding IP addresses to the moonraker whitelist, but it didn't help
I also did not understand how to find out the ip of the CYD itself
Help :c

suchmememanyskill, 2024-07-25:
Logs?

(You can get logs via the installer prompt, there's an option to open the terminal)

suchmememanyskill, 2024-07-25:
HTTPC_ERROR_CONNECTION_REFUSED

Btw, you can view the ip of cyd klipper on your router.

fylhtq7779, 2024-07-25:
> HTTPC_ERROR_CONNECTION_REFUSED
> 
> Btw, you can view the ip of cyd klipper on your router.

The problem is that I don't see this device in the router settings.
The message WiFi Status: Disconnected appears in the logs immediately after I connect to wifi

fylhtq7779, 2024-07-25:
> HTTPC_ERROR_CONNECTION_REFUSED
> 
> Btw, you can view the ip of cyd klipper on your router.

What does HTTPC_ERROR_CONNECTION_REFUSED mean?

fylhtq7779, 2024-07-25:
Hello again
I managed to fix it, but I'm not sure how
In the router settings, I set the standard DNS 8.8.8.8 and 8.8.4.4, and I was also able to find the MAC address and made it static

Thank you for your help and for your project!

closed 2024-07-25 by fylhtq7779