suchmememanyskill/CYD-Klipper #123: [Suggestion] Add ability to enter SSID with space

tit, 2024-08-08:
Hi!
My SSID name contains space, for example "Homer Simpson".
I tried to enter this name through serial monitor, but I get an error in the number of arguments.
```
> said Homer Simpson wifiPassWord
```

Please add ability to enter SSID with space.

suchmememanyskill, 2024-08-08:
This issue is specific to the serial console, correct?

suchmememanyskill, 2024-08-14:
Would 
> ssid Homer\ Simpson wifiPassWord
work for you?
(Backslash escapes next character)

bwucke, 2024-08-19:
@suchmememanyskill Since the console commands take a constant number of arguments, the escape would be best implemented in `tokenize` so that `ssid` still takes 2 arguments, just one of them contains space (after `tokenize` spaces are no longer a problem)

Want me to pick it up?

suchmememanyskill, 2024-08-19:
@bwucke check dev, already implemented the backslash thing (not very well, but it works)

closed 2025-01-10 by suchmememanyskill