suchmememanyskill/CYD-Klipper #126: ESP32-S3 support

yzheka, 2024-09-23:
Is it difficult to add ESP32-S3 board support?

MadRajib, 2024-11-30:
Hi @yzheka can you try https://github.com/MadRajib/CYD-Klipper-CrowPanel it uses  ESP32-WROOM-32-N4.

Also can you share the product link with that I can check the pinouts specific to this varient.

yzheka, 2024-11-30:
@MadRajib i will try today. Thank you. Here is the link https://a.aliexpress.com/_EJSKRcc

yzheka, 2024-12-02:
@MadRajib is not working. I can't upload, it says A fatal error occurred: firmware.bin is not an ESP32-S3 image. Also I have noticed in the task name it says 28R, I guess it is 2.8" resistive, but mine is 3.5" capacitive

MadRajib, 2024-12-02:
Hi @yzheka let me try locally to compile once for 3.5

dmit2k, 2024-12-09:
Hello!
Any success so far with this board?

MadRajib, 2024-12-10:
Sorry forgot to update the chat, I checked for the driver details it seems the tft lib doesn't support the driver for this display yet.