suchmememanyskill/CYD-Klipper #127: Setting for screen-off time (including not ever turning the screen off)

McNugget6750, 2024-10-01:
This is an amazing project. Works out of the box! Thank you so much! Just wanted to get this out of the way :)

I noticed that the screen has a timeout. I have the same on the Bambu xTouch but there, the timeout can be turned off.
It would be great if the timeout was configurable so that I can leave the screen on forever, if I want.

Cheers!
Timo

suchmememanyskill, 2024-10-01:
Out of curiousity, what would you usecase be for this? There's an option if your printer is printing, the screen stays on. So it'd only turn off while idle (which i think is fair, no use to keep it on if nothing is happening)

McNugget6750, 2024-10-01:
oh, I was going through the settings and didn't see the option. I must have missed it / my printer wasn't printing while I set up the display. If the option (while printing) already exists, we can close this issue! Thanks a lot!

closed 2024-10-01 by McNugget6750