suchmememanyskill/CYD-Klipper #130: Layer number not updating

btwebz, 2024-10-27:
Hi There!
Thank you for developing this great project. I am finding it incredibly useful. 
I am showing all statistics when printing however I have noticed that the 'Layer number of Layer number' is not updating.
It always just says 'Layer 0 of 0'. Have I missed something? I am runing the latest version of CYD-Klipper (v 1.8.0) with a SeeMeCNC Rostock V2.
Please let me know if I can provide any more information.
Thank you again.

suchmememanyskill, 2024-10-28:
Sounds like misconfiguration of your slicer settings. CYD-Klipper looks explicitly for https://docs.mainsail.xyz/overview/slicer/prusaslicer#layer-count

closed 2024-10-28 by suchmememanyskill

btwebz, 2024-10-28:
Awesome - thank you so much for your response!!