suchmememanyskill/CYD-Klipper #132: newest build. extrude/retract buttons dont work

KennyVB, 2024-11-03:
[image]
pushing on screen buttons results in this in console. klipper v0.12.0-339-g31fe50ffa

suchmememanyskill, 2024-11-03:
You seem to have enabled the option to use custom gcode for extrusion/retraction, while you have no macros defined for it

KennyVB, 2024-11-05:
> You seem to have enabled the option to use custom gcode for extrusion/retraction, while you have no macros defined for it

sadly, i havent done anything. just installed, and followed the guide to set it up= network, and then printer ip.
where is the option for disable/enable custom code?

suchmememanyskill, 2024-11-05:
The default option of 'Custom Filament Move Macros' is off. I don't know what else to say.
"Calls FILAMENT_RETRACT and FILAMENT_EXTRUDE in temperature menu when enabled".

Disable it, and it should use direct gcode commands for it again. You can set these items in settings (macros > settings > 'Custom Filament Move Macros')

suchmememanyskill, 2024-11-09:
Gonna asssume this is solved

closed 2024-11-09 by suchmememanyskill