suchmememanyskill/CYD-Klipper #135: ESP32-8048S043 screen is white after flashing

dmit2k, 2024-12-22:
Hello,

I have tried to flash my ESP32-8048S043 screen from the web but after flashing it goes into dead loop:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
...

Flashed successfully with Platformio, but the screen just goes white. 

From the serial terminal I was able to configure Wifi, Klipper IP and Moonraker key. But when tried to change the brightness to any value below 255 screen just goes black.

Also noticed that after some inactivity screen goes from full white to black, and touching the screen reverts it to full white state again. This makes me think touchscreen is at least working.

suchmememanyskill, 2024-12-22:
Sounds like you got a variant that is unsupported (probably wrong pinout for screen).
Sadly, i don't have this type of display

dmit2k, 2024-12-22:
Here is the link to the board documentation and examples from manufacturer:

http://pan.jczn1688.com/directlink/1/ESP32%20module/4.3inch_ESP32-8048S043.zip?spm=a2g0o.detail.1000023.1.6c2d7abZ7abZbG&file=4.3inch_ESP32-8048S043.zip

It has Arduino examples which I can compile and which work as expected. Maybe you or someone else could extract the correct parameters from this example Arduino code, please?

This example initialises LVGL and touch panel and as I can see has all required pinouts and parameters:
[3_3-4_TFT-LVGL-Widgets.zip](https://github.com/user-attachments/files/18223420/3_3-4_TFT-LVGL-Widgets.zip)

Thank you in advance for any help!

dmit2k, 2024-12-22:
I was able to make it work with the following changes to the board json:

        "'-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=14000000'",
        "'-D ST7262_PANEL_CONFIG_HSYNC_GPIO_NUM=39'",
        "'-D ST7262_PANEL_CONFIG_VSYNC_GPIO_NUM=41'",
        "'-D ST7262_PANEL_CONFIG_DE_GPIO_NUM=40'",
        "'-D ST7262_PANEL_CONFIG_PCLK_GPIO_NUM=42'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R0=45'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R1=48'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R2=47'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R3=21'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R4=14'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G0=5'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G1=6'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G2=7'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G3=15'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G4=16'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G5=4'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B0=8'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B1=3'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B2=46'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B3=9'",
        "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B4=1'",

The board has no additional hardware version indication, only ESP32-8048S043 on the PCB.

dmit2k, 2025-01-09:
I dug deeper into my issue and discovered that the incorrect colors were my mistake—there was no need to remap the pins!

The only parameter that needed fixing was ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ.

According to the manufacturer, there are no separate versions of this display. The correct value should be:

`
'-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=14000000'
`

When, why, and where the value (12.5*1000000) came from is a mystery, but most likely it was either:

    A workaround for some problematic batch from the very first production runs, where lowering the clocking was required for stable component operation.
    Or simply a copy-paste error from other sources about displays with slightly different hardware. And perhaps some devices happen to work with this frequency, which might be a case of "survivor bias."

I checked other projects where this display is used (including ESPHome), and everywhere the value is 14000000.

@suchmememanyskill, would you please fix this value in the template for esp32-8048S043C-SD (or at least add it as a newer board version)?

suchmememanyskill, 2025-01-09:
Would you mind submitting a pull request for this? (To the dev branch, please)

I can also edit it of course, if you wish

dmit2k, 2025-01-09:
> Would you mind submitting a pull request for this? (To the dev branch, please)

I could do this, but I think there are some decisions you should make yourself first as the project owner: 

1) according to the CYD-Klipper/boards/README.md:

"Contains specialised CYD definitions from [platformio-espressif32-sunton](https://github.com/rzeldent/platformio-espressif32-sunton)"

You get the boards defs from external source, so I'm not sure how should we properly patch them to not conflict with "original" version in the future

2) there is a chance that this old PCLK value is not a mistake and is correct for some older board batches, should we just create additional board config version (say esp32-8048S043C-smartdisplay-new.json)?

3) submissions to the dev branch can stay there for months before going to Master :)

suchmememanyskill, 2025-01-09:
I can make an alt version, sure.

If you mean the currently open PR, that's blocked by my own rewrite of the backend :)

suchmememanyskill, 2025-01-09:
I see that one has failed, oops, see https://github.com/suchmememanyskill/CYD-Klipper/commit/1cd6c8fd9208a05f7e05fb1dce0883487031052b

dmit2k, 2025-01-09:
WOW! I didn’t expect to trigger such an explosive reaction to a little trolling! :):):)
Awesome! My respect, tomorrow I’ll spend a long time testing it with pleasure! Thank you!

closed 2025-01-10 by suchmememanyskill