suchmememanyskill/CYD-Klipper #136: Gcodes subfolders not visible

mechano, 2025-01-02:
It's not able to navigate into gcodes subfolders.
I've configured a printer that when you insert an USB stick it uses udev to create a subfolder sda into printer_data/gcodes so it's possible to print from USB drive. 

But CYD doesn't see and doesn't surf into subfolders of printer_data/gcodes.

But also if I want to create subolders for different materials like ABS, ASA, PLA, TPU, PETG, I can't see them to select the file to print.

suchmememanyskill, 2025-01-02:
CYD-Klipper should show the full path of every file. 

Creating a folder structure using Moonraker's api is quite difficult. So i haven't implemented it. Instead, a full path is shown.

mechano, 2025-01-02:
So also the files into subfolders are shown and can be printed?

suchmememanyskill, 2025-01-02:
Yes, they should be able to.

Although, depends on the root of the path.

https://moonraker.readthedocs.io/en/latest/web_api/#list-available-files

Cyd-klipper only reads from the gcodes root.

suchmememanyskill, 2025-01-28:
I assume this issue is resolved/fixed?

closed 2025-01-28 by suchmememanyskill